dsmrreader/dsmr-reader #1250: InfluxDB en later field toevoegen

poudenes, 2021-01-09:
Hoi Allen,

Ik heb influxDB draaiende op DSMR. Alles gaat goed.. Alle data gaat netjes naar de db toe.
Ik heb later volgende toegevoegd in het scherm "InfluxDB-integratie wijzigen"

[electricity_cost]
electricity1_cost = electricity1_cost
electricity2_cost = electricity2_cost
electricity_cost_merged = electricity_cost_merged

[gas_cost]
gas_cost = gas_cost
total_cost = total_cost

Echter deze data komt niet in influxDB... 
Hoe kan ik deze data toch naar influxDB krijgen?

dennissiemensma, 2021-01-09:
Bedankt voor je vraag. Je kunt alleen de velden gebruiken die als voorbeeld in de configuratie staan. Dat zijn velden van de metingen zonder aggregatie.

De achterliggende gedachte van de InfluxDB-integratie is namelijk dat je de brondata (metingen) daar inzet zodat je daar je eigen aggregaties over kan doen. Kosten zijn geen onderdeel van de metingen (maar een laagje in DSMR-reader zelf) en zijn daardoor niet beschikbaar.

Het kan zijn dat die later nog beschilbaar komen via MQTT als generieke manier van datasynchronisatie.

dennissiemensma, 2021-01-10:
Ik zal de beschrijving in de configuratie bij de volgende release wat verduidelijken.

closed 2021-01-10 by dennissiemensma