dsmrreader/dsmr-reader #1251: Dagstatistieken lopen achter.

iepmaatje, 2021-01-12:
Hallo Allemaal, Ik draai DSMR met een aparte datalogger. Tot gisteren liep alles gesmeerd, maar vanochtend zag ik ineens de melding "   Dagstatistieken lopen achter   (1 dag, 12 uur geleden)"  Er is niets verandert door mij. Waar komt deze melding vandaag ? Ik heb nog steeds ' Live' data, alleen geen dagstatistiek van gisteren...

dennissiemensma, 2021-01-12:
Bedankt voor je vraag. Je kunt het beste eerst #1230 lezen en https://github.com/dsmrreader/dsmr-reader/issues/1197#issuecomment-724968648 proberen. Ik hoor graag of je daar iets kan vinden.

iepmaatje, 2021-01-12:
Hallo Dennis, ik lees de volgende Error:
" ERROR    HTTPSConnectionPool(host='data.buienradar.nl', port=443): Max retries exceeded with url: /2.0/feed/json (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f70d630b250>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
Traceback (most recent call last):"

Is het niet kunnen bereiken van Buienradar de oorzaak van het ontbreken van de dagstatistiek ?

dennissiemensma, 2021-01-12:
Bedankt voor de update. Je zult denk ik debug logging moeten aanzetten en zoeken naar de meldingen in de gelinkte comment in #1197 om de oorzaak te achterhalen. De buienradar error is ongerelateerd. En ik neem aan dat je geen andere meldingen hebt in de About pagina?

iepmaatje, 2021-01-13:
Hallo Dennis, Ik zie in de DEBUG dat de gas gegevens niet meer doorkomen.. dit zal de oorzaak zijn. Enig idee wat ik nu moet doen ? Sorry voor al mijn vragen...

2021-01-13 10:33:10,460] DEBUG    SP: Running "Generate day and hour statistics" (dsmr_stats.services.run)
[2021-01-13 10:33:10,486] DEBUG    Stats: Waiting for first gas reading on the next day...
[2021-01-13 10:33:10,488] DEBUG    SP: Rescheduled "Generate day and hour statistics" to 2021-01-13 10:38:10.486698+01:00 (ETA 0:04:59.998217)

iepmaatje, 2021-01-13:
Hallo Dennis, Na wat meer speurwerk zie ik dat mijn slimme meter geen gasgegevens ontvangt.. Ligt dus niet aan DSMR.
Tijdelijk uitzetten van de gasgegevens in de backend heeft vootlopig de oplossing gebracht. Thanks voor je support !

dennissiemensma, 2021-01-13:
Fijn dat je het gevonden hebt. Mocht je later nog andere issues hebben, koppel het gerust weer terug!

closed 2021-01-13 by dennissiemensma