dsmrreader/dsmr-reader #1252: dsmr_webinterface FATAL & OSError: [Errno 28] No space left on device

Raarrrinator, 2021-01-13:
Goedenavond,

Zojuist viel het me op dat de DSMR webinterface niet meer draait met als opmerking 'dsmr_webinterface FATAL'.

Mijn DSMR draait op een NUC in Virtualbox.

Een error die ik ook krijg is: OSError: [Errno 28] No space left on device

Eerst had de VDI 8gb ruimte, deze heb ik daarom vergroot naar 16gb.

[image]

De error blijft echter. In sommige posts hier las ik dat sudo rm /var/tmp/gunicorn--dsmr_webinterface.pid zou kunnen helpen. Dit bestand wordt echter niet gevonden. De bestanden in de map /var/tmp lijk ik niet te kunnen verwijderen (geen rechten? en zien er als volgt uit:

[image]

De database lijkt nog te lopen. Hebben jullie een idee hoe het op te lossen is? Bedankt!

Het volgende heb ik aan data weten te verzamelen via SSH:

`login as: pi
pi@192.168.1.62's password:
Linux raspberry 4.19.0-8-686 #1 SMP Debian 4.19.98-1+deb10u1 (2020-04-27) i686

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Jan 13 21:24:36 2021 from 192.168.1.9

pi@raspberry:~ $ sudo supervisorctl
dsmr_backend                     RUNNING   pid 1718, uptime 0:00:02
dsmr_datalogger                  STARTING
dsmr_webinterface                FATAL     Exited too quickly (process log may have details)

pi@raspberry:~ $ sudo supervisorctl tail -f dsmr_webinterface
==> Press Ctrl-C to exit <==
ast):
  File "/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn", line 8, in <module>
    sys.exit(run())
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 58, in run
    WSGIApplication("%(prog)s [OPTIONS] [APP_MODULE]").run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/base.py", line 228, in run
    super().run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/base.py", line 72, in run
    Arbiter(self).run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/arbiter.py", line 198, in run
    self.start()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/arbiter.py", line 137, in start
    self.pidfile.create(self.pid)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/pidfile.py", line 37, in create
    os.write(fd, ("%s\n" % self.pid).encode('utf-8'))
OSError: [Errno 28] No space left on device

pi@raspberry:~ $ sudo view /var/log/supervisor/dsmr_webinterface.log
[2020-12-22 22:24:41 +0100] [8491] [INFO] Starting gunicorn 20.0.4
[2020-12-22 22:24:41 +0100] [8491] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (8491)
[2020-12-22 22:24:41 +0100] [8491] [INFO] Using worker: sync
[2020-12-22 22:24:41 +0100] [8494] [INFO] Booting worker with pid: 8494
[2020-12-22 22:30:37 +0100] [8491] [INFO] Handling signal: term
[2020-12-22 22:30:37 +0100] [8494] [INFO] Worker exiting (pid: 8494)
[2020-12-22 22:30:37 +0100] [8491] [INFO] Shutting down: Master
[2020-12-22 22:31:04 +0100] [690] [INFO] Starting gunicorn 20.0.4
[2020-12-22 22:31:04 +0100] [690] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (690)
[2020-12-22 22:31:04 +0100] [690] [INFO] Using worker: sync
[2020-12-22 22:31:04 +0100] [918] [INFO] Booting worker with pid: 918
[2020-12-22 22:44:54 +0100] [690] [INFO] Handling signal: term
[2020-12-22 22:44:54 +0100] [918] [INFO] Worker exiting (pid: 918)
[2020-12-22 22:44:54 +0100] [690] [INFO] Shutting down: Master
[2020-12-22 22:44:59 +0100] [2363] [INFO] Starting gunicorn 20.0.4
[2020-12-22 22:44:59 +0100] [2363] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (2363)
[2020-12-22 22:44:59 +0100] [2363] [INFO] Using worker: sync
[2020-12-22 22:44:59 +0100] [2368] [INFO] Booting worker with pid: 2368
[2020-12-22 23:07:42 +0100] [2363] [INFO] Handling signal: term
[2020-12-22 23:07:42 +0100] [2368] [INFO] Worker exiting (pid: 2368)
[2020-12-22 23:07:42 +0100] [2363] [INFO] Shutting down: Master
[2020-12-22 23:08:07 +0100] [752] [INFO] Starting gunicorn 20.0.4
[2020-12-22 23:08:07 +0100] [752] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (752)
[2020-12-22 23:08:07 +0100] [752] [INFO] Using worker: sync
[2020-12-22 23:08:07 +0100] [918] [INFO] Booting worker with pid: 918
[2020-12-22 23:10:34 +0100] [752] [INFO] Handling signal: term
[2020-12-22 23:10:34 +0100] [918] [INFO] Worker exiting (pid: 918)
[2020-12-22 23:10:34 +0100] [752] [INFO] Shutting down: Master
[2020-12-22 23:21:51 +0100] [620] [INFO] Starting gunicorn 20.0.4
[2020-12-22 23:21:51 +0100] [620] [INFO] Listening at: unix:/var/tmp/gunicorn--dsmr_webinterface.socket (620)
[2020-12-22 23:21:51 +0100] [620] [INFO] Using worker: sync
[2020-12-22 23:21:51 +0100] [900] [INFO] Booting worker with pid: 900
[2020-12-22 23:39:10 +0100] [900] [INFO] Autorestarting worker after current request.
[2020-12-22 23:39:10 +0100] [900] [INFO] Worker exiting (pid: 900)
[2020-12-22 23:39:10 +0100] [1938] [INFO] Booting worker with pid: 1938
[2020-12-23 00:10:19 +0100] [1938] [INFO] Autorestarting worker after current request.
[2020-12-23 00:10:19 +0100] [1938] [INFO] Worker exiting (pid: 1938)
[2020-12-23 00:10:19 +0100] [3163] [INFO] Booting worker with pid: 3163
[2020-12-23 16:48:15 +0100] [3163] [INFO] Autorestarting worker after current request.
[2020-12-23 16:48:15 +0100] [3163] [INFO] Worker exiting (pid: 3163)
[2020-12-23 16:48:15 +0100] [9733] [INFO] Booting worker with pid: 9733
"/var/log/supervisor/dsmr_webinterface.log" [readonly] 877L, 58062C

pi@raspberry:~ $ cd /home/dsmr/dsmr-reader/
pi@raspberry:/home/dsmr/dsmr-reader $ sudo -u dsmr /usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--TEST.socket --pid /var/tmp/gunicorn--TEST.pid dsmrreader.wsgi
[2021-01-13 21:27:07 +0100] [1923] [INFO] Starting gunicorn 20.0.4
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn", line 8, in <module>
    sys.exit(run())
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 58, in run
    WSGIApplication("%(prog)s [OPTIONS] [APP_MODULE]").run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/base.py", line 228, in run
    super().run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/app/base.py", line 72, in run
    Arbiter(self).run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/arbiter.py", line 198, in run
    self.start()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/arbiter.py", line 137, in start
    self.pidfile.create(self.pid)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/gunicorn/pidfile.py", line 37, in create
    os.write(fd, ("%s\n" % self.pid).encode('utf-8'))
OSError: [Errno 28] No space left on device

pi@raspberry:/home/dsmr/dsmr-reader $ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            473M     0  473M   0% /dev
tmpfs           101M  4.3M   96M   5% /run
/dev/sda1       6.9G  6.7G     0 100% /
tmpfs           502M     0  502M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           502M     0  502M   0% /sys/fs/cgroup
tmpfs           101M     0  101M   0% /run/user/1000

pi@raspberry:/home/dsmr/dsmr-reader $ sudo supervisorctl restart dsmr_webinterface
dsmr_webinterface: ERROR (not running)
dsmr_webinterface: ERROR (spawn error)
`

dennissiemensma, 2021-01-13:
Bedankt voor je vraag. Ik weet het niet helemaal zeker, maar het kan zijn dat de Pi niet doorheeft dat er mee ruimte is. Je zou `sudo raspi-config` kunnen proberen en dan `6 Advanced options` > `Expend filesystem`. Al weet ik niet of dat ook direct de root-partitie vergroot, die nu vol zit.

Ik neem verder aan dat je geen retentie hebt ingesteld in DSMR-reader. Als de Pi straks weer werkt, raad ik je aan dat sowieso aan te zetten en na een paar dagen een vacuumdb te doen: https://dsmr-reader.readthedocs.io/en/latest/faq.html#how-do-i-reclaim-database-disk-space

Voor nu zul je eerst moeten kijken of je ergens wat ruimte kan krijgen. Mijn eerste ingeving zou zijn om wat backups weg te gooien in `/home/dsmr/dsmr-reader/backups/`. Dat zou je genoeg ruimte moeten geven om bovenstaande aanpassingen te doen. Doe daarna vooral ook even een Pi-reboot.

Raarrrinator, 2021-01-13:
> Expand filesystem

De optie was niet aanwezig in advanced options, daarom heb ik het zo geprobeerd: raspi-config --expand-rootfs

Met dit resultaat:

[image]

dennissiemensma, 2021-01-13:
Probeer vooral mijn laatste suggestie. Daarmee krijg je sowieso tijdelijk ruimte om retentie aan te zetten en de database te laten opschonen.

Raarrrinator, 2021-01-14:
> Probeer vooral mijn laatste suggestie. Daarmee krijg je sowieso tijdelijk ruimte om retentie aan te zetten en de database te laten opschonen.

Dankjewel; ik heb de backups en tmp files verwijderd. Daarna geprobeerd de vacuumdb te doen, maar ik vermoed dat er elders iets niet klopt. Ik denk dat ik DSMR opnieuw ga installeren op een nieuwe VM.

`
pi@raspberry:~ $ sudo su - postgres
postgres@raspberry:~$ vacuumdb -f -v -z -d dsmrreader
vacuumdb: could not connect to database dsmrreader: could not connect to server:                            No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
`

dennissiemensma, 2021-01-14:
Je zult eerst ruimte moeten maken. Die heb je nu tijdelijk. Dan kun je proberen om je hele systeem te rebooten. Daarmee zou dsmrreader en de DB weer moeten draaien. Vervolgens is de we interface weer beschikbaar en kun je retentie aanzetten. Als dat een paar uur of dagen draait, pas dan kun je een vacuumdb doen. Nu krijg je een foutmelding omdat de DB niet draait.

dennissiemensma, 2021-01-18:
@Raarrrinator ben je hier nog verder mee gekomen?

Raarrrinator, 2021-01-19:
> @Raarrrinator ben je hier nog verder mee gekomen?

Ik heb ruimte gemaakt en alles opnieuw opgestart. De DB en DSMR reader werkten, maar de interface bleef 'Fatal'. Het gekke is, is dat wanneer ik het VM venster open deze ook geheel zwart is; er draait wel iets, maar ook het Raspbian desktop is niet zichtbaar. Dus ik stel hem binnenkort opnieuw in in een nieuwe VM met snapshots de installs door.

To be continued!

dennissiemensma, 2021-02-16:
Mocht je hier nog ergens tegenaan lopen, laat maar weten.

closed 2021-02-16 by dennissiemensma