dsmrreader/dsmr-reader #1253: Nieuw URL voor de mindergas API

Erik050572, 2021-01-14:
Goedemiddag Dennis,

Ik zag op [Mindergas](https://mindergas.nl/member/api) dat ze een nieuwe URL hebben voor het uploaden van meterstanden.

> Meterstand API
>Wijzigingen per 1 januari 2021
>
>In verband met het toevoegen van ondersteuning voor meterstanden in kWh en GJ zijn de volgende wijzigingen aan de API doorgevoerd:
>
 >   De URL is veranderd van http(s)://www.mindergas.nl/api/gas_meter_readings naar http(s)://www.mindergas.nl/api/meter_readings.
>    De meterstand in liters (reading_l) wordt niet meer ondersteund.
>
> De oude API blijft voorlopig nog wel werken (zeker enkele maanden), maar update ajb jouw software.

Heb je hier al rekening meegehouden?  :)

Mvg,
Erik

dennissiemensma, 2021-01-14:
Bedankt voor je melding. Ik had dat niet meegekregen (of ik heb er een mailtje over gemist). Ik zal het wijzigen in de volgende release.

dennissiemensma, 2021-01-14:
Doorgevoerd voor volgende release.

closed 2021-01-14 by dennissiemensma

dennissiemensma, 2021-01-17:
Uitgebracht in [v4.11](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.11.0)