dsmrreader/dsmr-reader #1255: Nieuwe update in een MQTT meesturen

poudenes, 2021-01-15:
Hoi Allen,

Ik open regelmatig de DSMR pagina om te kijken of er een update is.
Het zou mooi zijn als er een MQTT meegezonden wordt wanneer er een nieuwe versie beschikbaar is.

Systeem checkt oude versie gelijk aan nieuwe versie dan is MQTT Update "false"
oude versie anders dan nieuwe  versie dan is MQTT Update "true"

Zo kan ik in HA bijvoorbeeld een automation maken. Een notificatie sturen als er een update beschikbaar is

:)

dennissiemensma, 2021-01-15:
Bedankt voor je verzoek. Wellicht dat ik het meeneem zodra er nieuwe MQTT-features komen, maar voorlopig laat ik het even voor wat het is.

Thuurke, 2021-01-27:
Ik los dit nu op door de notifications pagina van de installatie uit te lezen. Als hier There is a newer version of..dan wordt de counter meer dan 1. Opslaan in een json en die laten uitlezen. werkt naar behoren.

`import datetime as dt
from urllib.request import urlopen
import json

now = dt.datetime.now()
now_formatted = now.strftime("%d-%m-%Y %H:%M:%S")

url = "http://x.x.x.x/notifications"
page = urlopen(url)
html = page.read().decode("utf-8")
text_to_find = "There is a newer version of DSMR-reader available."
text_index = html.find(text_to_find)

count = html.count(text_to_find)

if count == 0: 
   status="Uit"
else:
   status="Aan"

datastore = []
datastore = {
             'status':     status,
             'count':     count,
             'last_checked': now_formatted,
             }

json.dumps(datastore)

with open("dsmr_update.json", "a") as outfile:
    json.dump(datastore, outfile)
    outfile.write('\n')    `

dennissiemensma, 2021-01-27:
@Thuurke je kunt nog overwegen om simpelweg de XHR-request te doen die de About-pagina ook doet:

```
GET /about/xhr/update-check
```

```
{"is_latest_version": true}
```

Wellicht versimpelt dat je check.

Thuurke, 2021-01-27:
@dennissiemensma  Oke dat zou inderdaad wat code regels schelen :)

skank01, 2021-02-11:
+1 voor update mqtt

dataegg, 2021-02-18:
> @Thuurke je kunt nog overwegen om simpelweg de XHR-request te doen die de About-pagina ook doet:
> 
> ```
> GET /about/xhr/update-check
> ```
> 
> ```
> {"is_latest_version": true}
> ```
> 
> Wellicht versimpelt dat je check.

Met deze info kun je hier prima een binary sensor voor maken in HA

dennissiemensma, 2021-03-07:
Ik heb hier opnieuw naar gekeken, maar het is relatief veel werk om dit via MQTT te realiseren, dus ik laat het voor wat het is. Dan besteed ik liever mijn beschikbare tijd aan features zoals #1291.

Ook omdat er alternatieve methodes zijn. 
- `GET /about/xhr/update-check`
- https://api.github.com/repos/dsmrreader/dsmr-reader/tags

closed 2021-03-07 by dennissiemensma