dsmrreader/dsmr-reader #1257: Energiecontracten

PhyxionNL, 2021-01-16:
Een aantal vragen/opmerkingen over energiecontracten.

1. **Totaal** staat onder het gas, dit is niet heel logisch omdat gas en stroom verschillende eenheden zijn. Eigenlijk zou je het zo willen: 
[image] 
Eventueel aangevuld met **Totaal** kWh net als de tarieven erboven.

2. Ik heb mijn retentieconfiguratie op "Uitgeschakeld" staan, dit wil ik graag op één jaar zetten, maar wat voor impact heeft dit op de energiecontracten? Er staat alleen:
_De volgende gegevens worden nooit opgeschoond:_
    Dagtotalen/statistieken (zoals weergegeven in het Archief).
    Uurtotalen/statistieken (zoals weergegeven in het Archief).
    De eerste meting van elk uur.
    De laatste meting van elk uur.

3. "Daarnaast wordt teruggeleverde energie niet verrekend in de kosten. De teruggeleverde kWh wordt hier alleen voor de volledigheid getoond.". Dat klopt niet want dit wordt wel meegerekend (wat overigens klopt want de opbrengst van de opgeleverde energie kan je uiteraard aftrekken van de totale kosten van je energiecontract).

dennissiemensma, 2021-01-16:
Bedankt voor je vragen. Hierbij per stuk een toelichting:

1. De huidige opzet is inderdaad niet ideaal, maar was meer om het compact te houden (vooral op kleinere schermen). Ik wil best wel een keer voor je kijken of ik er nog een regel tussen kan drukken, als dat het duidelijker maakt.

2. De [totalen van de energiecontracten](https://github.com/dsmrreader/dsmr-reader/blob/2c1ce8550b1ea9fd6e5ff769538b1cde85fa267e/dsmr_consumption/services.py#L500-L504) komen uit [dezelfde bron](https://github.com/dsmrreader/dsmr-reader/blob/2c1ce8550b1ea9fd6e5ff769538b1cde85fa267e/dsmr_stats/services.py#L280-L297) als het Archief, dus automatisch opschonen heeft daar **geen** invloed op. Eigenlijk werkt het hetzelfde als [het Archief](https://github.com/dsmrreader/dsmr-reader/blob/2c1ce8550b1ea9fd6e5ff769538b1cde85fa267e/dsmr_stats/services.py#L253-L277) in jaar-niveau, alleen zijn de start- en einddatums gebaseerd op de contractdatums (tenzij iemand toevallig ook contracten heeft van 1 jan - 31 dec).

3. Het is mij niet helemaal duidelijk wat je bedoelt. Heb je misschien een screenshot van je eigen situatie? Ik heb zelf geen teruglevering, dus ik kan hier niet makkelijk simuleren wat je bedoelt.

PhyxionNL, 2021-01-16:
Zie onderstaand 😊
1. Op zich hoeft dat niet eens, alleen het totaal één regel omhoog (evt. met kWh erachter) zoals op mijn screenshot is genoeg. Nu staat die onderaan en is het net alsof het gas er ook bij hoort.
2. Prima, dan pas ik dat aan.
3. In de omschrijving staat die tekst: 
[image] 
DSMR reader doet dat wel, en dat is ook correct (de oplevering gaat van het verbruik af, want dat kost je natuurlijk ook niks), alleen die tekst klopt dus niet. 

[image]

dennissiemensma, 2021-01-16:
Aah helder, dank voor de toelichting. Ik zal de tekst voor teruglevering wat aanpassen. Het ging mij er destijds vooral om dat DSMR-reader geen rekening houdt qua kosten met salderen (vooral bij afwijkende saldering). 

Wat betreft de totalen, ik weet dan niet of dat dan helemaal lekker gaat. Het was de bedoeling dat die totaalregel qua prijs het totaal van alles is en niet alleen van elektra. Vandaar dat er dan mogelijk een extra regel bij moet.

PhyxionNL, 2021-01-17:
> Wat betreft de totalen, ik weet dan niet of dat dan helemaal lekker gaat. Het was de bedoeling dat die totaalregel qua prijs het totaal van alles is en niet alleen van elektra. Vandaar dat er dan mogelijk een extra regel bij moet.

Dat lijkt mij ook handig inderdaad, maar een totaal van verschillende eenheden (zoals dat nu het geval is) is in ieder geval niet erg nuttig.

dennissiemensma, 2021-01-17:
Ik heb losgetrokken voor de volgende release:

[image]

[image]

dennissiemensma, 2021-01-17:
De tekst bovenin is straks ook aangepast en wat generieker gemaakt.

closed 2021-01-17 by dennissiemensma

dennissiemensma, 2021-02-17:
Onderdeel van [v4.12](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.12.0)

martijnb92, 2021-03-07:
Het lijkt mij persoonlijk handig om deze nieuwe layout ook toe te passen op de dashboard pagina ipv alleen op de energiecontracten pagina :smiley:

dennissiemensma, 2021-03-07:
Ik zal het meenemen voor de release van aankomende week. Dashboard is daar namelijk sowieso al wat aangepast:

[image]