dsmrreader/dsmr-reader #1258: Error after upgrading to V4

Jasper-GIT, 2021-01-16:
I'm running DSMR on Raspbian on a pi3. Is was running since 2018 perfect. After upgrading from V3 to V4 I get te error page below:
Also removed DSMR and installed again with keeping my database. But still the same issue.

```
Server Error

Sorry, something unexpected happened.
Exception:

ProgrammingError: column dsmr_stats_daystatistics.fixed_cost does not exist
LINE 1: ...daystatistics"."total_cost") AS "total_cost", SUM("dsmr_stat...
                                                             ^
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 11, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 98, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 159, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 35, in get_context_data
    target_date=timezone.localtime(timezone.now()).date()

  File "/home/dsmr/dsmr-reader/dsmr_stats/services.py", line 290, in month_statistics
    return range_statistics(start=start_of_month, end=end_of_month)

  File "/home/dsmr/dsmr-reader/dsmr_stats/services.py", line 273, in range_statistics
    temperature_avg=Avg('average_temperature'),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 398, in aggregate
    return query.get_aggregation(self.db, kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/sql/query.py", line 500, in get_aggregation
    result = compiler.execute_sql(SINGLE)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1156, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
```

dennissiemensma, 2021-01-16:
Bedankt voor je melding. De tweede foutmelding krijg je omdat de migraties nog niet voltooid zijn. Als ze blijven hangen dan zullen we moeten kijken waar het aan ligt:

- Hoelang heb je ongeveer gewacht totdat die je migraties hebt gebroken?
- Wat voor hardware gebruik je? (bijv: Pi 2, Pi 3, Pi 4)
- Wat is de output van dit command:
```
sudo su - dsmr
./manage.py showmigrations | grep -v '[X]'
```

Jasper-GIT, 2021-01-16:
Bedankt voor je snelle reactie.

De migratie loopt al 90 min. 
Ik gebruik een Pi3 (volgens mij de +)

output van je commando is:
`-su: ./manage.py: Bestand of map bestaat niet`

Wat heel gek is, want ik heb eerder het manage.py gebruikt en toen deed hij wel wat.

dennissiemensma, 2021-01-16:
Kan het zijn dat je er 1 regel van hebt gemaakt? Je kunt ze even apart proberen:

```
sudo su - dsmr

./manage.py showmigrations | grep -v '[X]'
```

Jasper-GIT, 2021-01-16:
Ik heb ze afzonderlijk uitgevoerd. 
Zie net nadat ik voor de zoveelste keer de migrate heb gestart dat hij nu klaar is en succesvol, maar ik heb geen grafische weergave van de webserver. Enkel een pagina met platte tekst op t moment. Hij begint nu wel iets meer te doen maar nog niet het gewenste

dennissiemensma, 2021-01-16:
Wat is de output van het commando? Het voert namelijk geen migraties uit, maar toont de status. Mogelijk is daarmee te zien waar die op wacht.

Jasper-GIT, 2021-01-16:
dit is de output van mijn migrate:

```
pi@raspberrypi:~ $ sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_mqtt.0015_continuous_client... OK
  Applying dsmr_mqtt.0016_client_settings_update... OK
  Applying dsmr_mqtt.0017_mqtt_keep_reconnecting... OK
  Applying dsmr_mqtt.0018_mqtt_always_reconnect... OK
  Applying dsmr_notification.0008_dummy_notification_provider... OK
  Applying dsmr_stats.0015_fixed_daily_cost... OK
```

dennissiemensma, 2021-01-16:
Oke dat is mooi. Wat komt hier nu uit: https://github.com/dsmrreader/dsmr-reader/issues/1258#issuecomment-761564293?

Jasper-GIT, 2021-01-16:
Nog steeds niets. 
Hier mn console:

```
pi@raspberrypi:~ $ sudo su - dsmr
dsmr@raspberrypi:~$ ./manage.py showmigrations | grep -v '[X]'
-su: ./manage.py: Bestand of map bestaat niet
dsmr@raspberrypi:~$ 
```

dennissiemensma, 2021-01-16:
Zit je wel in de goede map? Wat krijg je als je `pwd` typt daar?

Ergens tijdens de installatie is er een stukje script wat je aan de bashrc kan toevoegen, waardoor elke keer dat je `sudo su - dsmr` doet, deze automatisch dit doet:
```
source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader
```

Anders zul je dat elke keer handmatig moeten doen.

https://dsmr-reader.readthedocs.io/nl/v4/installation.html#virtualenv

Jasper-GIT, 2021-01-16:
Verkeerde folder dus:
```
(dsmrreader) dsmr@raspberrypi:~$ pwd
/home/dsmr
```

dit is nu mijn output:
```
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader$ ./manage.py showmigrations | grep -v '[X]'
admin
auth
contenttypes
dsmr_api
dsmr_backend
dsmr_backup
dsmr_consumption
dsmr_datalogger
dsmr_frontend
dsmr_influxdb
dsmr_mindergas
dsmr_mqtt
dsmr_notification
dsmr_pvoutput
dsmr_stats
dsmr_weather
sessions
```

dennissiemensma, 2021-01-16:
Mooi, dat klopt dan. Er staan nu geen migraties meer open. Dan kun je verder met waar je was gebleven met de upgrade of herinstallatie!

Jasper-GIT, 2021-01-16:
Nou, die is afgerond en ik krijg nu als ik naar de webserver ga de DSMR pagina in platte tekst zonder grafische weergave. Wat gaat daar dan nog mis?

dennissiemensma, 2021-01-16:
Als je dit uitvoert:
```
sudo su - dsmr

./deploy.sh
```

Krijg je dan ergens foutmelding?

En anders kun je dit proberen:
```
sudo su - dsmr

./manage.py collectstatic --noinput
```

Jasper-GIT, 2021-01-16:
Die eerste deed het. Nu werkt het, bedankt voor je hulp!

dennissiemensma, 2021-01-16:
Top, fijn om te horen!

closed 2021-01-16 by dennissiemensma