dsmrreader/dsmr-reader #1259: Update naar v4.10.0 lukt niet

mthmulders, 2021-01-16:
In de webinterface zie ik dat er een update beschikbaar is:

[image]

Dan log ik met SSH in en volg de gebruikelijke stappen voor het uitvoeren van een upgrade:

```
sudo su - dsmr
./deploy.sh
```

De uitvoer eindigt dan met:

    
    
     --- Clearing cache...
    
    
     --- Deployed version: 
    4.9
    
    
     >>> Deployment complete. <<<
    
    

Ik heb dit een paar keer geprobeerd maar ik blijf te zien krijgen dat 4.9 uitgerold is.

dennissiemensma, 2021-01-17:
Kan het zijn dat er een foutmelding in de output staat? En anders kun je de complete output delen.

mthmulders, 2021-01-17:
Hmmm, deze komt voorbij:

    --- Merging/updating checkout.
    Updating 044c8127..2c1ce855
    error: Your local changes to the following files would be overwritten by merge:
            dsmrreader/config/defaults.py
    Please commit your changes or stash them before you merge.
    Aborting

Ik heb de wijzigingen in *dsmrreader/config/defaults.py* ongedaan gemaakt en nu slaagt de deployment wel. Dat moet de oorzaak geweest zijn.

Interessant detail: ondanks dat de merge naar v.4.10.0 niet lukt gaat het script wel door. Deze fout stond vrij vroeg in de uitvoer, maar het script gaat door en geeft uiteindelijk geen duidelijk signaal dat één van de stappen mislukt is.

Los van zelf zorgvuldig de logs inspecteren (wijze les voor de volgende keer) zou het ook wel mooi zijn als het deploy script zelf signaleert dat er waarschijnlijk iets niet in orde is.

dennissiemensma, 2021-01-17:
Bedankt voor je terugkoppeling. Zo'n check lijkt me een goede toevoeging. Ik zal kijken of ik iets kan toevoegen.

dennissiemensma, 2021-01-17:
Aangezien je bezig was met updaten, er is nu ook [v4.11](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.11.0)

dennissiemensma, 2021-01-17:
Vanaf de v4.12 release zit er straks een extra check in:

```
 --- Checking for local file changes: 
 M deploy.sh

[!] Local file changes detected, aborting deployment
    To revert all files to their default, WITHOUT backup, execute:

    git reset --hard HEAD
```

closed 2021-01-17 by dennissiemensma

dennissiemensma, 2021-02-17:
Onderdeel van [v4.12](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.12.0)