dsmrreader/dsmr-reader #1261: InfluxDB: Field type conflict: input field “returned_2” is type float, already exists as integer

SanderGit, 2021-01-17:
DSMR-reader: 4.10 in Docker

Ik heb deze aan InfluxDB gekoppeld. Er komt wel data in terecht maar het gaat niet goed. Als ik de data handmatig wil synchroniseren komt er een foutmelding.

Debug info:
```
DSMR-READER
    App / Python / Database                                                       v4.10 / v3.9.1 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              24097
    Consumption records electricity / gas (est.)                                                   2609 / 309

```
Foutmelding:

```
raise InfluxDBClientError(err_msg, response.status_code)                                           
influxdb.exceptions.InfluxDBClientError: 400: {"error":"partial write: field type conflict: input field \"returned_2\" on measurement \"electricity_positions\" is type float, already exists as type integer 
dropped=200"}                                                                     

```
Het weggooien en opnieuw aanmaken van de InfluxDB database helpt niet.

dennissiemensma, 2021-01-17:
Bedankt voor je melding. Voor de duidelijkheid: Krijg je ook een fout in de automatische synchronisatie? 
Of alleen bij het uitvoeren van `./manage.py dsmr_influxdb_export_all_readings`?

SanderGit, 2021-01-17:
> Bedankt voor je melding. Voor de duidelijkheid: Krijg je ook een fout in de automatische synchronisatie?
> Of alleen bij het uitvoeren van `./manage.py dsmr_influxdb_export_all_readings`?

Bij het configureren van de automatische sync krijg ik geen melding. Als er een issue met de verbinding is gaat het vinkje ‘enabled’ uit. Een foutmelding heb ik niet gezien. De pagina ‘Influxdb measurements’ is leeg sinds de koppeling er in principe wel is.

Er wordt volgens mij wel data in geschreven, alleen krijg ik het er in Grafan nog niet uit met dit dashboard:
https://github.com/iotassistant/iotassistant.io/blob/master/grafana-dashboard.json

dennissiemensma, 2021-01-17:
De pagina `Influxdb measurements` in DSMR-reader weergeeft alleen gegevens die wachten op verzending. Als er geen achterstand is, zul je daar inderdaad niet vaak gegevens zien.

Krijg je dezelfde fout als je dit probeert?
```
./manage.py dsmr_influxdb_export_all_readings --to-influx-database testje
```
./manage.py dsmr_influxdb_export_all_readings --to-influx-database testje

SanderGit, 2021-01-17:
Dan doet ie het zonder foutmelding.

dennissiemensma, 2021-01-17:
Je kunt nog proberen om het eerst te exporteren naar een aparte InfluxDB-database, met bovenstaand commando (totdat alles over is). En daarna ook de automatische export te wisselen naar die database.

SanderGit, 2021-01-17:
Het is nog weinig data dus ik vind het wel goed zo. Ik zal dit issue sluiten. Bedankt voor het meedenken!

Wel ben ik benieuwd of er ergens een lijst met variabelen/velden is die ik naar InfluxDB kan sturen, ik vind in de MQTT-documentatie wel wat dingen maar heb nog geen compleet overzicht gezien.
Power failures, sags e.d. zie ik er in de standaardconfiguratie niet standaard bij staan bijvoorbeeld.

closed 2021-01-17 by SanderGit