dsmrreader/dsmr-reader #1265: database op andere postgreSQL server

Simanias, 2021-01-19:
Ik wil dmsr-reader via een Proxmox LXC Ubuntu 18.04 container installeren. Ook heb ik een RancherOS VM in Proxmox draaien waarop postgreSQL middels docker draait, als database voor home assistant. Dit werkt super, de latency is echt top.

De installatie van dsmr-reader gaat goed maar ik zou graag deze postgreSQL server die ik ook voor home assistant gebruik, ook gebruiken voor dmsr-reader (waarom weer een postgreSQL instance laten draaien, als ik al een postgreSQL server heb). 

Ik heb de settings.py als volgt aangepast (komt uit dit topic: #395)
```
"""
    For advanced users only, override any other settings (unavailable as env var) by defining them at the end of this file.
"""
from dsmrreader.config.production import *

# Add any overrides below this line and restart/reload the application with: ./reload.sh
DATABASES = {
    'default': {
        'ENGINE': 'django.db.backends.postgresql',
        'NAME': os.environ.get('DB_NAME', 'dsmrreader'),
        'USER': os.environ.get('DB_USER', 'dsmrreader'),
        'PASSWORD': os.environ.get('DB_PASS', '*********'),
        'HOST': os.environ.get('DB_HOST', '192.168.1.13'),
        'PORT': 5432,
        'CONN_MAX_AGE': os.environ.get('CONN_MAX_AGE', 'none'),
    }
}
```

De database _dsmrreader_ heb ik aangemaakt in met pgAdmin in postgreSQL en gevuld met oude data, dit is succesvol. Ook heb ik een dsmrreader user aangemaakt met een password ******. De postgreSQL server draait op 192.168.1.13:5432, dit werkt aangezien home assistant er ook gebruik van maakt.
Als ik het ./manage.py migrate commando uitvoer, gaat het mis. Volgens mij doe ik iets niet goed met settings.py, maar ik heb geen idee.

Heeft iemand een idee of suggestie om het bovenstaande voor elkaar te krijgen?

dennissiemensma, 2021-01-19:
Je kunt het beste de bovenstaande config weer terugzetten naar wat het was. Het issue dat je linkt is zwaar outdated. Tegenwoordig kun je je instellingen doen via `.env` in de root van het project of via env vars (Docker).

Alle mogelijke opties staan hier: https://dsmr-reader.readthedocs.io/en/latest/env_settings.html#django-settings-overrides

Simanias, 2021-01-26:
@dennissiemensma dankjewel voor de referentie.
Op dit moment heb ik er voor gekozen om dsmr-reader middels docker te laten draaien. Dit omdat ik het niet voor elkaar kreeg om de serial2usb interface, middels een pass-through te laten herkennen in de LXC container.
Middels de env variabelen heb ik het geheel aan de praat gekregen. Er draait nu een dsmr-reader database op de postgreSQL server, waarop ook home assistant draait.
Hieronder de instellingen in portainer:

DJANGO_TIME_ZONE | Europe/Amsterdam
-- | --
DJANGO_DATABASE_HOST | 192.168.1.13
DJANGO_DATABASE_USER | dsmrreader
DJANGO_DATABASE_PASSWORD | *******
VIRTUAL_HOST | localhost
.....
DJANGO_SECRET_KEY | dsmrreader
DJANGO_DATABASE_ENGINE | django.db.backends.postgresql
DSMRREADER_SUPPRESS_STORAGE_SIZE_WARNINGS | True
DJANGO_DATABASE_NAME | dsmrreader
DJANGO_DATABASE_PORT | 5432
DSMRREADER_ADMIN_USER | admin
DSMRREADER_ADMIN_PASSWORD | admin
DATALOGGER_MODE | standalone
SD_LOGLEVEL | info
SD_USER | root
SD_GROUP | root

Ik ben niet weg van de manier om een docker container te updaten en aangezien er best vaak een nieuwe release van dsmr-reader uitkomt weet ik niet of ik docker manier acceptabel vind.
Dank voor de info.

dennissiemensma, 2021-01-26:
Bedankt voor je update.

> Dit omdat ik het niet voor elkaar kreeg om de serial2usb interface, middels een pass-through te laten herkennen in de LXC container.

Je zou ook nog de andere kant op kunnen proberen, door de remote datalogger te gebruiken. Dan hoef je alleen een HTTP-poort naar die container te hebben. De remote datalogger stuurt het dan via de API van DSMR-reader.

Of het proberen via `ser2net`, wat je seriele poort over netwerk beschikbaar maakt. Al zul je daarvoor wel `ser2net` ergens moeten draaien waar die toegang tot de seriele poort heeft. 

> Ik ben niet weg van de manier om een docker container te updaten en aangezien er best vaak een nieuwe release van dsmr-reader uitkomt weet ik niet of ik de docker manier van dsmr-reader draaien, acceptabel vind.

Ik kan me voorstellen dat het relatief veel werk lijkt voor kleinere updates. Echter is het goed om te realiseren dat major updates (zoals 5.x) wel in 1x goed gaan, zonder dat je daar zelf ook maar enige upgrade-guide voor hoeft te volgen. Dat is immers al voor je gedaan in de Docker image. Dat kan nog een mogelijke overweging voor je zijn.

Simanias, 2021-01-27:
> Je zou ook nog de andere kant op kunnen proberen, door de remote datalogger te gebruiken. Dan hoef je alleen een HTTP-poort naar die container te hebben. De remote datalogger stuurt het dan via de [...]

Ja op dit moment draai ik dsmr-reader binnen VMWare esxi en heb ik aparte raspberry als remote datalogger, dit werkt prima. Echter ik vind het 'mooier' om alles op de server proberen te laten draaien (gekke tik).

dennissiemensma, 2021-01-27:
Helder. Mocht je nog meer informatie nodig hebben, laat het gerust weten.

closed 2021-01-27 by dennissiemensma

Simanias, 2021-01-27:
@dennissiemensma toch nog een update, ik weet dat het buiten de scope van dsmr-reader valt, maar het is uiteindelijk met veel werk toch gelukt om het in een LXC container te laten draaien en de serial2usb middels een pass-through te laten herkennen in de container. De database draait op postgreSQL op een VM in RancherOS.
Mocht iemand meer details willen weten, PM.
Nogmaals dank voor de info en succes met dsmr-reader!