dsmrreader/dsmr-reader #1267: Is de FAQ ook afgestemd voor Docker?

MRL-33, 2021-01-23:
Momenteel heb ik DSMR Reader lopen in een container.
Ik heb zitten te zoeken in de FAQ, maar ik denk dat er geen uitleg is hoe je DSMR of database kunt upgraden wanneer je DSMR Reader als container hebt werken.
Momenteel werk ik met versie 4.9 met een database met postgres:10-alpine (zie docker-compose.yaml)

[image]

1) Ik heb geprobeerd om DSMR Reader te upgraden door de container te stoppen en te verwijderen en daarna de docker-compose.yaml opnieuw te starten.
2) Voor het upgraden van de datebase: ik weet niet hoe dat moet.
3) Voor het wissen van de database: Ik kan deze niet vinden op mijn Raspberry 3B+

Zou je mij hiermee willen helpen?

Mijn bestand: /home/pi/docker-compose-files/dsmr/docker-compose.yaml

`version: '3.8'

# the dashboard is reachable at: http://<HOSTNAME>:7777
# volume: folder_on_host:folder_inside_container

services:
  dsmrdb:
    image: postgres:10-alpine
    container_name: dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmrdb:/var/lib/postgresql/data
    environment:
      - TZ=${TZ}
      - PG_TZ=${TZ}
      - POSTGRES_USER=${POSTGRES_USER}
      - POSTGRES_PASSWORD=${POSTGRES_PASSWORD}
      - POSTGRES_DB=${POSTGRES_DB}
#    networks:
#      - net-dsmr

  dsmr:
    image: xirixiz/dsmr-reader-docker:latest
    container_name: dsmr
    restart: always
    depends_on:
      - dsmrdb
    cap_add:
      - NET_ADMIN    
    links:
      - dsmrdb
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmr_backups:/dsmr/backups
    environment:
      - TZ=${TZ}
      - DJANGO_TIME_ZONE=${TZ}
#      #Nginx related:
#      - ENABLE_NGINX_SSL=false  # Enables port 443 for nginx (/etc/ssl/private/fullchain.pem and /etc/ssl/private/privkey.pem are required to be mounted!)
#      #Nginx .httpassword (thanks @Caroga):
#      - ENABLE_HTTP_AUTH=false
#      - HTTP_AUTH_USERNAME=
#      - HTTP_AUTH_PASSWORD=
#      - VIRTUAL_HOST=127.0.0.1
#      - VIRTUAL_PORT=80
#      - SD_AUTOSTART_MQTT=true
#      - SD_AUTORESTART_MQTT=true
#    networks:
#      - net-dsmr
    ports:
      - 7777:80
      - 7779:443
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0

#networks:
#  net-dsmr:
#    driver: bridge
    
#volumes:
#   dsmrdb:
#   dsmrdb_backups:
`

dennissiemensma, 2021-01-23:
Bedankt voor je vragen. Ik kan me voorstellen dat het wat verwarrend is. Het is goed om te weten dat ik zelf beperkt kennis van Docker heb en ook de opzet voor Docker niet beheer. Deze worden beheerd in een apart project.
Alle credits voor het uitwerken (en vooral onderhouden) van de Docker-versie van DSMR-reader gaat uit naar @xirixiz (en voorheen @ualex73). 

Aan je config te zien gebruik je de setup van Xirixiz. Ik denk dat je het beste in zijn project [de README kan doornemen](https://github.com/xirixiz/dsmr-reader-docker#dsmr-reader---releases-info) en als je daar de nodige info niet kan vinden, even [door oude issues heen zoeken](https://github.com/xirixiz/dsmr-reader-docker/issues?q=is%3Aissue+is%3Aclosed). Als je er dan nog niet uitkomt, kun je altijd daar nog een issue aanmaken.

Ik vermoed dat je hier wat meer info kan vinden:
- https://github.com/xirixiz/dsmr-reader-docker/issues/149
- https://github.com/xirixiz/dsmr-reader-docker/issues/152

_Het lijkt erop dat je gewoon de laatste image kan gebruiken. Die zou de DB-structuur (waar nodig) moeten upgraden. Er vanuitgaande dat je doelt op de structuur in de DB en niet de PostgreSQL-versie zelf._

ualex73, 2021-01-23:
De container upgraden is eigenlijk zoals alle andere containers, image tag aanpassen en docker run of docker-compose up gebruiken. Eigenlijk wordt er niets in de DSMR container bewaard.

De PostgreSQL upgraden is eigenlijk zoals de export/import is, maar daar tussenin doe je dan een docker run/docker-compose up zoals je bij de DSMR container deed :-) Dus tussen stap 2 en 3 in mijn lijstje:
https://hub.docker.com/repository/docker/ualex73/dsmr-reader-docker

xirixiz, 2021-01-24:
Het is zoals @ualex73 al aangeeft. Je kan de tag aanpassen van de container voor DSMR naar de laatste versie en daarmee draait DSMR op de laatste release. Maar aangezien je de latest tag gebruikt kan je ```docker-compose pul dsmrl``` of  ```docker pull xirixiz/dsmr-reader-docker:latest``` voor de zekerheid uitvoeren. Eigenlijk, zoals je het beschrijft zou Docker inderdaad de laatste images op moeten halen, mits je ```docker-compose up``` en ```docker-compose down``` gebruikt. Alleen bij het stoppen en starten van een container gebeurd er niets, dan wordt gewoon alleen hetgeen al beschikbaar/bestaat opnieuw gestart.

Ik zou overigens eerst de upgrade van dsmr uitvoeren en pas daarna de database upgrade.

Een tweede stap is de database upgraden. De data van verise 10 werkt niet direct in versie 12, dus hierbij geldt het niet dat alleen een tag naar de laatste release postgres upgrade. Wanneer je naar Postgres versie 12 gaat moet de data ook bruikbaar zijn voor Postgres 12. De volgende stappen zou je uit kunnen voeren wanneer je er niet uitkomt met het voorbeeld van ualex73: https://github.com/xirixiz/dsmr-reader-docker#upgrade-the-postgres-container-to-a-newer-release

Volgens mij voldoende info om verder te komen! Succes!

dennissiemensma, 2021-01-24:
Dank voor jullie bijdrages! @MRL-33 heb je hiermee genoeg informatie?

MRL-33, 2021-01-24:
@dennissiemensma , @xirixiz , @ualex73 
Dank voor de snelle hulp.
Met [https://hub.docker.com/r/ualex73/dsmr-reader-docker](url) lees ik joe ik de Postgres-database kan upgraden naar laatste Postgres-versie
E.e.a. wordt me nu duidelijker:

A) Export database:
export data (Postgres versie 10) naar SQL,
B) Restore Datebase:
stap 2: Wis oude database
stap 3: maak nieuwe Postgres-database (versie 12)
stap 4: import SQL-data in Postgres-database (versie 12)

Toch nog een paar vragen voordat ik dit alles uitvoer:
- Maar wanneer moet ik mijn docker-compose.yaml bijwerken?
- Stap 3 is mij niet geheel duidelijk mbt Postgres-versie. Maakt "docker exec -t db-dsmr createdb -O dsmrreader dsmrreader -U dsmrreader" nu de nieuwste Postgres-database versie 12 aan?

Mijn bestand: /home/pi/docker-compose-files/dsmr/docker-compose.yaml
`version: '3.8'

#the dashboard is reachable at: http://<HOSTNAME>:7777
#volume: folder_on_host:folder_inside_container

services:
  dsmrdb:
    image: postgres:10-alpine
    container_name: dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmrdb:/var/lib/postgresql/data
    environment:
      - TZ=${TZ}
      - PG_TZ=${TZ}
      - POSTGRES_USER=${POSTGRES_USER}
      - POSTGRES_PASSWORD=${POSTGRES_PASSWORD}
      - POSTGRES_DB=${POSTGRES_DB}
#networks:
#- net-dsmr

  dsmr:
    image: xirixiz/dsmr-reader-docker:latest
    container_name: dsmr
    restart: always
    depends_on:
      - dsmrdb
    cap_add:
      - NET_ADMIN    
    links:
      - dsmrdb
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmr_backups:/dsmr/backups
    environment:
      - TZ=${TZ}
      - DJANGO_TIME_ZONE=${TZ}
#      #Nginx related:
#      - ENABLE_NGINX_SSL=false  # Enables port 443 for nginx (/etc/ssl/private/fullchain.pem and /etc/ssl/private/privkey.pem are required to be mounted!)
#      #Nginx .httpassword (thanks @Caroga):
#      - ENABLE_HTTP_AUTH=false
#      - HTTP_AUTH_USERNAME=
#      - HTTP_AUTH_PASSWORD=
#      - VIRTUAL_HOST=127.0.0.1
#      - VIRTUAL_PORT=80
#      - SD_AUTOSTART_MQTT=true
#      - SD_AUTORESTART_MQTT=true
#    networks:
#      - net-dsmr
    ports:
      - 7777:80
      - 7779:443
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0

#networks:
#  net-dsmr:
#    driver: bridge
    
#volumes:
#   dsmrdb:
#   dsmrdb_backups:`

dennissiemensma, 2021-01-25:
Je kunt wellicht het beste je vraag bij Xirixiz zelf leggen, gezien je zijn image gebruikt. Ook zodat anderen de vragen en antwoorden later kunnen teruglezen: https://github.com/xirixiz/dsmr-reader-docker/issues

Als je de stappen van Ualex wilt gebruiken, kun je het beste wisselen naar zijn image. Ik weet namelijk niet of je de stappen en images door elkaar heen kan gebruiken.

xirixiz, 2021-01-25:
Ik ben het eens met @dennissiemensma. 

Stap 0: stop en verwijder de dsmr docker container
Stap 1: export de db data van pg10
Stap 2: stop en verwijder de pg10 docker container
Stap 3: verwijder de data van de db (locatie volume mount)
Stap 4: pas docker-compose aan met pg12
Stap 5: start pg12 via docker-compose
Stap 6: importeer de export van stap 1
Stap 7: start dsmr via docker-compose

Werkt dit niet, maar dan idd een issue aan bij ualex73 of bij mij voor verdere ondersteuning.

closed 2021-01-26 by dennissiemensma