dsmrreader/dsmr-reader #1268: Onduidelijkheid over berekening kosten

gigatexel, 2021-01-25:
Het wordt in de documentatie niet duidelijk vermeld hoe de energiekosten precies berekend worden.

In de Belgische energiemarkt is het zo dat alles gebaseerd is op kwartieren, zowel verbruik als tarief op basis van kwartierpiek (zie ook https://github.com/dsmrreader/dsmr-reader/issues/1084).

Dit heeft ook impact op de berekening van de kosten voor afname en injectie.

Voorbeeld:
Van 12u00 tot 12u15 is er:

- 5kWh afname
- 1kWh injectie

Aanrekening op factuur: 4kWh afname

Van 12u00 tot 12u15 is er:

- 3kWh afname
- 4kWh injectie

Aanrekening op factuur: 1kWh injectie

Binnen 1 kwartier wordt dus 1 keer kosten (of bonus) aangerekend.

Hoe gebeurt dit in DSMR reader? Is dit ook op kwartierbasis?
Dit is voor de Vlaamse gebruikers een key feature, omdat alle aanrekeningen in het energiesysteem voor particulieren zodanig zijn opgezet.
Indien dit in DSMR reader niet zo gebeurt en ook niet zo configureerbaar is, zou dit duidelijk aangegeven moeten worden omdat dit tot verkeerd gedrag (en kosten) kan leiden.

closed 2021-01-25 by gigatexel

dennissiemensma, 2021-01-25:
Bedankt voor je vraag. Het 'klopt' dat DSMR-reader hier geen rekening mee houdt. #1084 staat pas op de planning zodra dat gaat spelen. Ik begreep dat dat niet eerder is dan 2022.

Wat betreft kosten, ik denk niet dat daar tzt ondersteuning voor komt mits het niet direct mapt op een kale kWh prijs. Dat zeggende hebben moet ik later dit jaar eerst maar eens een werkend opzet van #1084 hebben draaien om te kijken welke informatie wel beschikbaar is.
 Het wordt naar verwachting hoe dan ook een aparte pagina of onderdeel, dus wellicht is er wel iets te rekenen, zolang het maar simpel blijft. Want alle prijzen binnen DSMR-reader zijn puur bedoeld als indicatie, om iemand een gevoel te geven hoeveel een dag kost (bijv winter tov zomer, qua gas of teruglevering).

dennissiemensma, 2021-01-25:
Had je het issue trouwens bewust weer gesloten? Ik kan er eind dit jaar best een blik op werpen na #1084, maar dus onder voorbehoud.

gigatexel, 2021-01-26:
Dag Dennis,

Ik had het bij het verkeerde eind. Ik werk zelf bij de netbeheerder en blijkbaar is er geen compensatie op kwartierbasis. De kosten worden dus "ogenblikkelijk" verrekend. Als de meter 1kWh afname registreert, is dat 1kWh in kosten. Idem voor opbrengst bij injectie.
Is dit ook zo geïmplementeerd in DSMR-Reader? Het kwartier-verhaal voor facturatie is dus een fout van mezelf.

Ivm het piekvermogen, dat gaat in voege vanaf 01/01/2022. Ik zet in de oorspronkelijke PR ook mijn MySQL query die je misschien als basis kan gebruiken.

VBP8501, 2021-01-26:
@gigatexel : bedoel je dat de kwh prijs verschillend zal zijn doorheen de dag? 

Hoe ik het heb begrepen is dat je kwh prijs over heel het jaar vast ligt ifv je energiecontract. Vanaf 2022 is er het capaciteitstarief dat een deel van de distributie, wat nu in de kwh prijs wordt verekend, op een kW basis zal verrekenen (volgens de details in #1084). 

Daarnaast heb je nog een aparte kWh prijs voor injectie. 

Dus vanaf 2022 heb je:
- € / kWh hoeveelheids afname (deel voor productie, deel voor transport, deel voor distributie, ...)
- € / kW vermogen afname (volgens #1084)
- € / kWh injectie

Voor integratie in het dsmr dashboard is het op zich niet zo verschillend aan nu. Alleen de prijs voor het tweede punt hierboven opnemen. 

Geheel ter info voor wie de situatie in België niet zo goed kent: injectieprijzen nu liggen rond de 4 cent / kWh. De terugdraaiende teller (saldering denk ik dat het in NL noemt) is vanaf nu niet meer van toepassing door een vonnis van de rechtbank.

gigatexel, 2021-01-26:
De prijs per kWh hangt af van je contract en het type contract. Al decennia bestaan "vaste" contracten met vaste prijs voor zowel elektriciteit als gas gedurende 1 of meerdere jaren.
Al even lang bestaan ook contracten met een variabele prijs, waarbij de prijs elke maand wordt aangepast aan de markt. In de tijd van de analoge draaischijfmeter, werd je jaarverbruik via statistieken (SLP, synthetic load profiles) verdeeld over de maanden van de afgelopen facturatieperiode en zo gekoppeld aan de juiste maandprijs. Met de digitale meter wordt dat dan een exacte mapping tussen kWh en maandprijs.

Om de berekening in DSMR-reader goed te krijgen, zou je in DSMR moeten kunnen een leverancier en type contract kiezen en dan de juiste tarieven van de respectievelijke leverancierswebsites scrapen. Er bestaat volgens mij geen open API met energieprijzen voor de particuliere markt.
Zonder bovenstaande functie is de kostberekening in DSMR een goeie gok maar ook niet meer dan dat. Met een variabel contract zou je zelfs elke maand je prijs moeten aanpassen in de configuratie.

Vanaf 01/01/2021 bestaat er ook een tarief injectie, waarbij je een aantal eurocent krijgt per geïnjecteerde kWh. Die is dus "ogenblikkelijk" en niet gecompenseerd op minuut of kwartierbasis. Elke kWh die je afneemt of injecteert wordt direct gefactureerd.

Vanaf 01/01/2022 wordt het factuurdeel distributiekosten anders berekend. Op vandaag is dat een vaste term (euro/jaar) en een variabele term. Vanaf die datum wordt een extra factor in rekening gebracht afhankelijk van je kwartierpiek. Hoe hoog deze zullen zijn en hoe de berekening dan verloopt is nog niet duidelijk. Wordt dit een aantal euro proportioneel aan de kwartierpiek (eur/kWpiek/maand) of een factor waarmee de andere termen vermenigvuldigd worden? De minister en de VREG moeten daarover nog tot een vergelijk komen.
Opgelet: op deel injectie betaal je geen distributievergoeding en wordt ook niet meegenomen in de piekberekening. 

@dennissiemensma de juiste verrekening kan nog niet geïntegreerd worden aangezien die nog niet duidelijk is. Wat wel kan is het weergeven en archiveren van de maandpieken (op kwartierbasis, zie queries in https://github.com/dsmrreader/dsmr-reader/issues/1084)

VBP8501, 2021-01-26:
Je kan nu al de geldigheid van een contract met zijn prijs beperken tot 1 maand dus de variabele contracten zou je op die manier kunnen ingeven. Scrapen lijkt mij onbegonnen werkt gezien het nu al bijna onbegonnen werk is om van een leverancier zijn prijzen op zijn website te vinden en daarna die tool te gaan onderhouden. Misschien dat de api van dsmr-reader toelaat om contracten aan te maken. Dan kun je wel zelf een scraper maken voor jouw leverancier en die integreren met dsmr reader.

Dat de introductie van het capaciteitstarief in de factuur nog niet duidelijk is, wist ik niet. Dus het is niet noodzakelijk €/kW. Een factor om de rest mee te verhogen zou het wel helemaal zijn.. Ik geraak met moeite onder de 4kW piek. Als 2,5kW de basis is dan...

dennissiemensma, 2021-01-26:
Bedankt voor alle extra informatie. De huidige prijsberekening in DSMR-reader is vrij recht-toe-recht-aan: https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_consumption/services.py#L252-L284

Het Nederlandse gebruikers hebben meerdere malen gevraagd om extra prijsvelden of calculaties daarvoor toe te voegen, zodat het met hun energienota overeenkomt. Dat is geen doen, daar begin ik niet aan en heb ik telkens afgewezen. Ook al zijn de ideeen of suggesties nog zo tof. En DSMR-reader is ook te gebruiken zonder prijzen te gebruiken.
Dus zaken zoals het inbouwen van leveranciers en hun prijzen, valt wat mij betreft echt buiten de scope van dit project. Alle prijzen binnen DSMR-reader zijn puur bedoeld als indicatie, om iemand een gevoel te geven hoeveel een dag kost (bijv winter tov zomer, qua gas of teruglevering). Ik denk dat ik dat nog wat duidelijker moet vermelden in DSMR-reader zelf.

Echter, een endpoint in de DSMR-reader API voor contracten is dan wel weer een mogelijkheid. Al weet ik niet of dat alles voor iedereen oplost.

Ik denk dat ik later dit jaar in Q3/Q4 maar eens opnieuw alles ga doorlezen en om meer informatie vraag tegen die tijd. Zoals ik het voor me zie komt er straks een instelling (of detectie) in DSMR-reader, waardoor er een aparte pagina komt met kwartierpieken. Plus nog de bijbehorende dataopslag.
En als het echt tegenvalt dan heroverweeg ik de gehele feature tzt. Want ik voeg graag ondersteuning toe voor niet-Nederlandse situaties, maar dat moet wel werkbaar blijven natuurlijk. Dat is het grootste gevaar bij dit soort projecten (merk ik afgelopen jaren). :]

gigatexel, 2021-01-27:
Dag Dennis,

Voor een goed begrip, ik denk dat niemand hier verwacht dat DSMR Reader automatisch tarieven zal gaan scrapen, inlezen en verrekenen. Dat is ver buiten de scope van dit project denk ik :-)