dsmrreader/dsmr-reader #1269: Mogelijkheid 'verbergen' graphics

bennybubble, 2021-01-25:
Hallo Dennis,

Sinds kort maak ik gebruik van je software, installatie op een RPI3 ging bijna moeiteloos en alles draait naar behoren.

Bij het weergeven van de 'live graphs' staat bovenaan de 'recent electricity consumption' en daaronder de distributie per fase.
In mijn geval heb ik wel 3 fases aan de ingang (zoals zo velen) maar slechts één fase in gebruik, de tweede grafiek is dus een kopie van de eerste en voegt weinig toe.

Zou er een mogelijkheid zijn om deze 'uit te zetten' of the disablen ?

-ben

dennissiemensma, 2021-01-25:
Bedankt voor je melding! DSMR-reader toont (als het goed is) alleen de fasen-grafiek wanneer er daadwerkelijk meerdere fasen gedetecteerd zijn.

Kan het zijn dat je toevallig een 3-fasen meter hebt met slechts 1 fase aangesloten? Die situatie heb ik namelijk nog niet eerder gezien. Bij mij werd pas een 3-fasen meter geplaatst toen ik daadwerkelijk er om vroeg (nodig had), los van of de leidingen er al lagen. Dat is overigens wel 6 jaar geleden.

bennybubble, 2021-01-25:
Hallo Dennis,

> Bedankt voor je melding! DSMR-reader toont (als het goed is) alleen de fasen-grafiek wanneer er daadwerkelijk meerdere fasen gedetecteerd zijn.
> 
> Kan het zijn dat je toevallig een 3-fasen meter hebt met slechts 1 fase aangesloten? 

Ik denk het, heb niet naar het type van de meter gekeken maar weet wel dat er maar 1 fase in gebruik is (zoals ik reeds beschreef), spanningsmetingen geven waarden voor 3 fasen maar de energie is ( in mijn geval) alleen op fase 3 af te lezen.
-ben

dennissiemensma, 2021-01-26:
Bedankt voor je aanvulling. Ik denk dat je dan 'pech' hebt gehad dat die niet op fase 1 is aangesloten. Daar gaat DSMR-reader namelijk momenteel van uit. Ik kan er op termijn wel een instelling voor maken, maar dat zal wel even duren.

----

Ik denk dat je voor nu het meeste geholpen bent door zelf bij je eigen installatie de grafiek weg te halen. Je hoeft dan alleen dit blok weg te halen volgens mij: https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_frontend/templates/dsmr_frontend/live-graphs.html#L68-L85

Dus regel 68 t/m regel 85 in:
```
/home/dsmr/dsmr-reader/dsmr_frontend/templates/dsmr_frontend/live-graphs.html
```

Let wel dat je daar bij de eerstvolgende update een conflict in krijgt. Omdat het allemaal onder versiebeheer staat is dat overigens makkelijk te herstellen. Ook als je zometeen er een typefout in maakt en het niet goed krijgt. In beide gevallen kun je het volgende uitvoeren om alle lokale wijzigingen ongedaan te maken:

```
sudo su - dsmr
git reset HEAD --hard
```

bennybubble, 2021-01-26:
Hallo Dennis,

Bedankt,  ik ga er even naar kijken.
Het is niet ongebruikelijk dat ik een andere fase heb dan mijn buren, dit werd (en wordt) gedaan om alle fasen gelijk te belasten.

-ben

dennissiemensma, 2021-01-26:
Ik laat deze even open staan. Anders verdwijnt die later uit zicht.

dennissiemensma, 2022-02-28:
@bennybubble is het mogelijk dat je een telegram deelt?

Je kunt deze vinden in DSMR-reader op de URL: `/admin/dsmr_datalogger/meterstatistics/`

Je mag deze regels met je serienummer(s) weg laten:
- `0-0:96.1.1(.....`
- `0-1:96.1.0(.....`

closed 2022-07-05 by dennissiemensma