dsmrreader/dsmr-reader #1270: Schaal X-as bij live grafieken

gigatexel, 2021-01-27:
De live grafieken zijn geweldig, maar de schaal op de X-as is een beetje vervelend.
Het zou goed zijn om steeds een vaste cadans (bvb afgerond op 5min) en met vaste tussenafstand (opnieuwe bvb 5min)
Nu ziet het er wat vervelend en onnatuurlijk uit.
[image]

dennissiemensma, 2021-01-27:
Bedankt voor je suggestie. Ik begrijp je voorkeur helemaal, echter wordt het voor mij erg lastig om iedereen's eigen voorkeur te ondersteunen.

Het is goed om te weten dat er verschillende schermbreedtes zijn en daarnaast ook mobiele gebruikers. Verder zijn er gebruikers die elke seconde een datapunt hebben en anderen (zoals ik) elke minuut. Anderen hebben een dusdanig hoge datalogger sleep dat de interval tussen de datapunten weer heel groot is (meer richting jouw wens). Daarnaast zijn er gebruikers die alleen het laatste uur willen zien. Anderen willen weer alleen de laatste paar minuten. En een enkeling wilde de complete afgelopen week zien (met gevolgen voor render-performance). 

Al deze zaken hebben invloed op de X-as en de schaal daarvan. Kortom, het is niet te doen, dus de grafieken schalen nu simpelweg automatisch op basis van:
- Aantal datapunten (groeperen per minuut of niet, heeft de grootste invloed)
- Initiele 'zoom' (het grijze tijdlijn-blokje onderaan de grafieken) die in de admin in te stellen is op en bepaald percentage (ik dacht standaard 10%).

Wellicht kun je daar nog mee spelen om je eigen voorkeur te benaderen.

dennissiemensma, 2021-01-27:
Dat zeggende hebben kun je natuurlijk ook zelf wat spelen met de grafieken-library. Alle huidige scripts daarvoor vind je in je lokale DSMR-reader installatie: https://github.com/dsmrreader/dsmr-reader/tree/v4/dsmr_frontend/static/dsmr_frontend/js/dsmr-reader/live

De library is eCharts en die heeft vele opties: https://echarts.apache.org/en/index.html

Als je om wat voor reden je lokale wijzigingen wilt terugdraaien, voer dan dit uit:
```
sudo su - dsmr
git reset HEAD --hard
```

dennissiemensma, 2021-01-27:
Verder heb je nog de mogelijkheid (als alternatief of extra verrijking) om de aggregaties zelf te doen, buiten DSMR-reader, in bijvoorbeeld een InfluxDB/Grafana, zodat je helemaal de controle hebt over hoe data gegroepeerd en weergegeven wordt. DSMR-reader kan de bronmetingen exporteren naar InfluxDB (of iets anders via MQTT). 

Strikt gezien is DSMR-reader simpelweg een van de vele manierer om DSMR-data te visualiseren, maar het is uiteraard niet **de** manier. Voor iedereen zal dat anders liggen.

closed 2021-01-27 by dennissiemensma

gigatexel, 2021-01-27:
Dag dennis, misschien heb je me verkeerd begrepen of misschien ook niet.

Het is duidelijk dat je niet voor iedereen goed kan doen en dat niet alles zomaar configureerbaar kan/moet zijn via de interface.
Ik had het louter over de "ticks". Waar je ticks zet heeft eigenlijk geen impact op welke data je laat zien, of hoeveel of het zoomlevel.

dennissiemensma, 2021-01-27:
Bedankt voor je aanvulling. Het hangt uiteindelijk allemaal met elkaar samen. Want iemand die geen metingen groepeert, wil ze bijvoorbeeld niet per 5 minuten zien maar per seconde. Ik kan daar helaas geen onderscheid in maken. Daar doel ik met het hele verhaal op. 😁