dsmrreader/dsmr-reader #1271: Evaluatie thuisbatterij

VBP8501, 2021-01-29:
Dit project vormt een prachtige databron om te evalueren of een thuisbatterij een goede investering is. Je weet wat er binnenkomt en wat buitengaat. Je kan gebruik maken van de samengevoegde minuutwaarden om te berekenen of een batterij wordt opgeladen of kan worden ontladen. 

Gezien de Belgische situatie waarbij de terugdraaiende teller is afgeschaft, is het belangrijk om zoveel mogelijk van het verbruik zelf te produceren. Een batterij helpt hierbij door over een korte periode productie te bufferen. De periode is afhankelijk van de grootte van de batterij maar zal in normale situaties niet meer dan 1 dag productie kunnen bufferen. Maar wat is onder de omstandigheden dan meest economische grootte van de batterij?

Ik heb voor mijzelf een script gemaakt om de energiekost in mijn situatie te berekenen adhv de dsmr database. En wou dat met jullie delen:
https://gitlab.com/mathias.vanden.auweele/dsmrbatteryanalyser

Dit issue mag uiteraard afgesloten worden want het is niet mijn bedoeling om dit te integreren in DSMR reader. Maar als er mensen op zoek gaan naar een berekening dan kan hiernaar verwezen worden, of komen ze hier misschien automatisch op terecht.

closed 2021-01-29 by VBP8501

dennissiemensma, 2021-01-29:
Bedankt voor het delen en erg leuk om te horen dat dit project je heeft geholpen bij de berekening!

Mocht je hier nog anderen mee willen helpen, dan kun je overwegen je project eventueel actief delen op diverse fora. Ik ben namelijk bang dat dit issue hier op Github vrij snel verdwijnt in de backlog van de honderden oude issues van dit project. En er lezen ook maar een handvol mensen mee hier helaas.

VBP8501, 2021-01-29:
Ja dat klopt, ik ben inderdaad nog van plan om het actief te promoten op andere fora. Heb vandaag nog wat tips gekregen om de berekeningen efficienter te maken.