dsmrreader/dsmr-reader #1272: Importeren s

VincentA-github, 2021-01-30:
Ik heb onlangs DSMR opnieuw moeten installeren.
Ik heb nog een backup van de dag en uur statistieken in mijn mail in dit bestandstype “ dsmrreader-postgresql-partial-backup-2020-09-17.sql”
Hoe kan ik die na een herinstallatie importeren?
Ik kon er op de site en via Google niet echt iets over terugvinden

```

DSMR-READER
    App / Python / Database                                                       v4.11 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              28514
    Consumption records electricity / gas (est.)                                                   2597 / 480

```

dennissiemensma, 2021-01-30:
Bedankt voor je vraag. Ik heb even door de oude issues gezocht, maar kan het niet zo snel vinden.

Omdat je al een nieuwe installatie hebt draaien, is de oude backup in de huidige vorm niet te importeren. Dit komt omdat er gebruik gemaakt wordt van oplopende ID's en er dan botsing ontstaat.

Dit betekent echter niet dat je niets met de back-up kan. De oplossing is namelijk om de ID-kolom + ID-waardes uit de backup te halen. Hiervoor zul je er echter wat handig in moeten zijn maar ik denk dat het ook goed te automatiseren is, om handmatige fouten te voorkomen. 
Daarom zul je denk ik even moeten wachten totdat er een feature voor komt. Daarmee kunnen de gegevens met terugwerkende kracht geimporteerd worden. Ondertussen kun je gewoon je nieuwe installatie laten doorlopen.

dennissiemensma, 2021-01-30:
Toch nog het oude issue kunnen vinden. Dit is een alternatief om met de hand te doen: https://github.com/dsmrreader/dsmr-reader/issues/1145#issuecomment-703837670

Maar als je dat niet aandurft dan zou ik wachten op de automatisering hiervan.

VincentA-github, 2021-01-30:
Bedankt! Ik ga kijken of het me lukt en anders wacht ik op de automatisering ervan.

dennissiemensma, 2021-02-24:
@VincentA-github ik heb hier nog naar gekeken, maar het is uiteindelijk technisch te complex om hiervoor iets te maken. Helemaal omdat dit (nog maar) incidenteel nodig is.
Ik denk dat je het beste dit kan proberen https://github.com/dsmrreader/dsmr-reader/issues/1145#issuecomment-703837670

Als je dat niet ziet zitten of niet lukt, dan mag je de partial ook naar me toe mailen (github at dennissiemensma punt nl), dan zal ik die stappen voor je uitvoeren en de gestripte SQL naar je terugsturen. Die kun je dan in je database plakken. Uiteindelijk is dat hoe dan ook sneller.

VincentA-github, 2021-02-25:
Ok, thx!

closed 2021-02-25 by VincentA-github

dennissiemensma, 2021-02-25:
Is het je al gelukt dan? Laat vooral even weten als je er zelf niet uit komt.

dennissiemensma, 2021-03-07:
@VincentA-github is het je zelf gelukt?

VincentA-github, 2021-03-07:
Hoi Dennis,

Ik heb er eerlijk gezegd nog niet naar gekeken. Toch bedankt voor de navraag, heel tof.