dsmrreader/dsmr-reader #1273: Datalogger blijft herstarten bij nieuwe rp4 install

DennisVredeveld, 2021-02-04:
heb mijn nieuwe pi4 ingericht vanaf scratch volgens https://dsmr-reader.readthedocs.io/en/v4/installation.html#option-b-install-dsmr-reader-manually-explained
Ik ondervind echter twee problemen (die wellicht gerelateerd zijn). Als ik controleer dat alle drie de processen draaien, lijkt de datalogger om de zoveel seconden automatisch te herstarten.
In de logging zie ik het volgende herhaaldelijk voorbij komen: 
`[2021-02-04 19:46:51,305] ERROR    dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Trace
back (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 46, in read_telegram
    len(buffer)
RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 2641`
Wanneer ik vervolgens handmatig ga testen met cu, krijg ik steeds dit soort meldingen:
`(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ cu -l /dev/ttyUSB0 -s 9600 --parity=even -E q
Connected.
(
 eTVE3dR-rvVg/NoVm6,lYZ2-r-V]H9rrvvVVV,v6VVVV%g/Rm6,2mdZr-R6-r2%!E!%N/Rm6,2mdZr-Vlv ,6%!E!%N/Rm6l2mdZrmvvvv6mV22%!E!%N/Rm6l2mdZrmvvvv6m6r6%!E!%NoVm6,lYZr2mvmvv6R%g/Rm6,2-2mvmvvv6-vd%!E%N/Rm6l2-2mvmvvv6mv6!%NoVm6,2m6mvmV,,lZv6%!E%N2lr2-2mv-V%goV-2,lYZ2mv-6v$vVVvvV6vV6V,VV6R%goV-2lr2-6mvmRr6rvRRlvvvVemvVemYVe-RemV-2lr2m2-Re-BF%No-2vvdmdV%goV-2lr2m2(
    eTVE3dR-rvVg/NoVm6,lYZ2-rV]Y92rrvvVVV,v6VVVV%g/Rm6,2mdZr-R6-r2%!E!%N/Rm6,2mdHr-VlvdmY,Y%!E!%N/Rm6l2mdZrmvvvv6mVb2%!E!%N/Rm6l2mdZrmvvvv6m6r6%!E!%NoVm6,lYZr2mvmvv6R%g/Rm6,2-2mvlvvv6-vd%!E%N/Rm6l2-2mvmvvv6mv6%!E%NoVm6,2m6mvmV,,lZv6%!E%FoVm6,lYZ34'v-R%goVm6,lYZ6-r-%NoVm6,lYZ6mv-%NV-2lr2-2mv-V%goV-2
                                                                     lHZ2mv6v$vVVvvV6v&V6V,VV6R%goV-2lr2-6mvmRr6rvR2Ve-RemV2lr2m2-Re-BV%go-2vvdmdV%goV-2lr2m2cu: Got hangup signal

Disconnected.`
De disconnect komt soms direct en soms na een aantal berichten pas en de data zelf zou volgens mij (ben nieuw met DSMR) ook leesbaarder moeten zijn. 
Mijn meter is een Is Iskra MT 382, dus vandaar dat ik met `cu -l /dev/ttyUSB0 -s 9600 --parity=even -E q` getest heb maar andere baudrate of parity geven hetzelfde effect.
Zoals gezegd ben ik een nieuwe gebruiker van DSMR dus wellicht zie ik iets simpels over het hoofd maar ik ik weet vrij zeker de handleiding exact gevolgd te hebben en heb daarbij geen afwijkingen geconstateerd.

dennissiemensma, 2021-02-04:
Bedankt voor je melding. De eerste melding zie je omdat DSMR-reader iets van je meter terugkrijgt, maar er geen herkenbaar telegram uit kan halen. Je `cu` output bevestigt dat.

Kan het toevallig zijn dat je in Luxemburg woont en dat P1-poort versleuteld is (https://github.com/dsmrreader/dsmr-reader/issues/928#issuecomment-602245840)? Dat zou in ieder geval kunnen verklaren waarom er geen herkenbaar telegram uit je meter lijkt te komen. 

En herkenbaar telegram ziet er bijvoorbeeld zo uit: https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/tests/datalogger/test_example_dsmr50.py#L19

DennisVredeveld, 2021-02-04:
Bedankt voor de snelle reactie!
Nee, ik woon gewoon in Nederland en mijn meter is 'gewoon' door Enexis geplaatst.

dennissiemensma, 2021-02-04:
Zolang er geen bruikbare gegevens uit de meter komt kan ik je helaas beperkt helpen en is het voor mij ook [rondzoeken op Tweakers](https://gathering.tweakers.net/forum/list_messages/1578510?data%5Bfilter_keywords%5D=Iskra%20MT%20382). Wat voor kabeltje gebruik je?

DennisVredeveld, 2021-02-04:
Ok bedankt, ik snap dat als er geen bruikbare data uit komt, DSMR er ook niet veel aan kan doen :)
Zal eens kijken of ik m.b.v. tweakers wijzer kan worden.
Ik gebruik deze kabel: [https://www.amazon.nl/Domoticz-Raspberry-Universele-P1-kabel-Kamstrup/dp/B089YLC5DT/ref=asc_df_B089YLC5DT/?tag=nlshogostdsp-21&linkCode=df0&hvadid=470481573771&hvpos=&hvnetw=g&hvrand=10569592807056449750&hvpone=&hvptwo=&hvqmt=&hvdev=m&hvdvcmdl=&hvlocint=&hvlocphy=9065068&hvtargid=pla-1028116665567&psc=1]
Aangezien mijn meter daar expliciet in de beschrijving genoemd wordt, ga ik er vanuit dat dat zou moeten werken.

dennissiemensma, 2021-02-04:
Ik neem aan dat je deze ook al geprobeerd had bij je eerdere pogingen?
```
cu -l /dev/ttyUSB0 -s 9600 --parity=none
```

DennisVredeveld, 2021-02-04:
Yep, zelfde resultaat:
```dsmr@raspberrypi:~/dsmr-reader $ cu -l /dev/ttyUSB0 -s 9600 --parity=none
Connected.
(
 eԓ?E?dR????Vg???V???l?Z2???]?92rrvvVVV,v????֒V?֒??%g/Rm6,2mdZr-R??6-r?%!š?ίR??,2mdZr-VlVl-6?%!E!%N/Rm6l2mdZ?mvvvv6mV22%!E!?N/Rm6l2mdZrmvv??6m6r6%!E!%NoVm6,lYZr2mvmvv6R%g/R???2-2??????6m66%!E%N/Rm6l2-2m?mvvv6mv6!%NoVm6,2m6mvmV,,lZ??%!E???V???l?Z??'v-R%goVm6,lYZ6-r-%NoVm6,lYZ6mv-%NoV-2lr2-2m?$V%goV-2,lYZ2m?-6v$vVVvv֒?????֒???,V?֒?ҥg?V???r2-6mvmRr6rv2rvv????????m??????V??lr2m2???BF??o-2vVlmvvV%goV-?lr2m2mv-R%g/+Vcu: Got hangup signal

Disconnected.
```

dennissiemensma, 2021-02-04:
Je zou daarnaast nog een berichtje kunnen proberen naar je netbeheerder om te vragen waar het aan ligt, al weet ik niet of ze je hiermee kunnen/willen helpen.

dennissiemensma, 2021-02-05:
Vergeet trouwens niet om in de DSMR-reader dataloggerconfiguratie (de admininterface) de datalogger op DSMR v2/v3 te zetten. Mocht je dat nog niet gedaan hebben. Dit verandert niet de output van `cu`, maar is wel nodig voor DSMR-reader zelf. [Dat is onderwater `9600 7E1`](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/services/datalogger.py#L41-L44).

 Ik zie nu dat ik ook al een keer een telegram van iemand had ontvangen [met dezelfde meter](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_datalogger/tests/datalogger/test_iskra.py#L25-L46) (lang lang geleden zelfs: #120). Alleen staan daar niet de seriele instellingen bij.

DennisVredeveld, 2021-02-05:
Dat is een goede, dat had ik inderdaad nog niet ingesteld!
9600 7E1 is ook in mijn post op tweakers bevestigd (zie https://gathering.tweakers.net/forum/view_message/66016646), dus dat betekent toch dat ik met cu moet testen met `-s 9600 parity=even` of moet ik de 7 databits ook nog expliciet opgeven?
In dat antwoord wordt ook over inverteren m.b.v. FT_Prog, dus ik ga dat eerst maar even uitzoeken. Helaas zie ik daar dan weer niks over op http://domoticx.com/p1-poort-slimme-meter-hardware/

dennissiemensma, 2021-02-05:
Ik zou eerst proberen of het nu misschien direct werkt in DSMR-reader. Je zult het datalogger-proces (of alle DSMR-reader processen) wel even moeten herstarten voor de zekerheid.
 
Ik weet niet welke argumenten nog nodig zijn voor `cu` om 7E1 te bereiken.

DennisVredeveld, 2021-02-06:
Finally solved! Het probleem zat het inderdaad in het feit dat bij mijn meter het RX-signaal kennelijk geinverteerd is en de kabel daardoor verkeerd uitleest. Door m.b.v. FT_Prog de kabel correct in te stellen (RX inverteren), komt er nu wel leesbare data uit en werkt DSMR reader ook meteen!
Wellicht handig mochten mensen na mij ooit hetzelfde probleem ervaren.
@dennissiemensma  in ieder geval bedankt voor je snelle en uitgebreide hulp, vooral ook omdat het niets te maken had met de reader zelf!

dennissiemensma, 2021-02-06:
Bedankt voor je update en toelichting. Fijn om te horen dat het nu werkt!

closed 2021-02-06 by dennissiemensma

dennissiemensma, 2021-02-06:
@DennisVredeveld nog 1 vraag, met welke `cu` parameters kun je nu de meter uitlezen?

In de DSMR-reader docs staat nu deze voor DSMR2-meters:
```
cu -l /dev/ttyUSB0 -s 9600 --parity=none
```

Maar als die niet klopt dan pas ik hem aan voor een volgende lezer.

DennisVredeveld, 2021-02-08:
> @DennisVredeveld nog 1 vraag, met welke `cu` parameters kun je nu de meter uitlezen?
> 
> In de DSMR-reader docs staat nu deze voor DSMR2-meters:
> 
> ```
> cu -l /dev/ttyUSB0 -s 9600 --parity=none
> ```
> 
> Maar als die niet klopt dan pas ik hem aan voor een volgende lezer.

Klopt helemaal!