dsmrreader/dsmr-reader #1274: How to change the web port?

rbeniers, 2021-02-05:
Hello,

I tried to change the standard web port (80) to another port number.

I opened the dsmr-webinterface file to change the port number using the command : nano /etc/nginx/sites-enabled/dsmr-webinterface.
I changed the port number to another one in the following line:
server {
    listen      80;
    server_name _;  # No hostname available? Just use the underscore for wildcard matching. :]

And try to save the new file. However I am not able to change the file. I get the next message: [ Error writing /etc/nginx/sites-enabled/dsmr-webinterface: Permission denied ].

I tried to edit the file with the root and the DSMR user, but both were not successfully.

How can I change the port in this file?

dennissiemensma, 2021-02-05:
Je zult het bestand waarschijnlijk moeten openen als sudo-gebruiker. 

Als je ingelogd bent als `pi` gebruiker zou `sudo nano /etc/nginx/sites-enabled/dsmr-webinterface` afdoende moeten zijn.

dennissiemensma, 2021-02-05:
Als root-gebruiker zou je trouwens nooit issues moeten hebben met permissies. Ook daar kun je proberen er `sudo` voor te zetten.

rbeniers, 2021-02-05:
Dat helpt: 'sudo' ervoor zetten. Thanks!

closed 2021-02-05 by dennissiemensma