dsmrreader/dsmr-reader #1275: Som gasverbruik

FutureCow, 2021-02-07:
Om een beetje inzicht te kunnen krijgen in het gasverbruik, zou het mooi zijn als je bijvoorbeeld kunt zien hoeveel gas er wordt verbruikt om het huis op temperatuur te krijgen. Het makkelijkste is als je bij de Live Grafieken de som zou zien van wat er op dat moment in beeld is. Als je dus van 6:00 tot 8:00 selecteert in de balk eronder, dat er ergens het getal staat wat alle staven bij elkaar opgeteld zijn, zodat je snel kunt vergelijken.

Zoiets:
[image]

Daarnaast nog een ander "kleine" vraag. Soms kom ik er na 3 dagen achter dat door een update DSMR niet meer draait. Nu moet ik zelf met de hand gaan uitrekenen hoeveel ik die dagen heb verbruikt en invoeren. Dit zou automatisch moeten kunnen door oude en nieuwe meterstand te vergelijkingen en dit over de missende dagen te verdelen?

dennissiemensma, 2021-02-07:
Bedankt voor je suggestie. Ik zal in een toekomstige versie kijken wat er mogelijk is qua grafieken, dat hangt namelijk erg af van de grafiekentool die ik gebruik.

dennissiemensma, 2021-02-07:
Wat betreft je tweede vraag is het denk ik handiger om de oorzaak van falende updates te achterhalen, zodat het uberhaupt niet meer voor komt. Heb je enige indicatie wat dat falen binnen DSMR-reader veroorzaakt?

dennissiemensma, 2021-03-22:
Ik heb hier wat dieper naar gekeken, echter lijkt het erop dat de grafiekentool in DSMR-reader het niet ondersteunt. Er is een hoop mogelijk, maar ik kon helaas geen manier vinden om op basis van de zoom het totaal op te halen: 

https://echarts.apache.org/examples/en/index.html

closed 2021-03-22 by dennissiemensma