dsmrreader/dsmr-reader #1276: Installatiescript werkt niet naar behoren met huidige repo v4

mvklingeren, 2021-02-07:
Ik probeer op een raspberry-pi (raspberry os), schone installatie DSMR te installeren. Echter ontbreekt het postgresql.txt bestand in de repo clone. 
Volgende regel wordt daardoor niet goed uitgevoerd (de base.txt deps worden niet geinstalleerd) > wat de installatie een drama maakt, mocht je dat niet door hebben ;-)

pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt

dennissiemensma, 2021-02-07:
Bedankt voor je melding. Kan het zijn dat je een oudere versie van de docs gebruikt? 

Voor DSMR-reader v4 zul je ook de `v4` (of nu nog `latest`) moeten gebruiken:
- https://dsmr-reader.readthedocs.io/en/v2/installation.html
- https://dsmr-reader.readthedocs.io/en/v3/installation.html
- https://dsmr-reader.readthedocs.io/en/v4/installation.html
- https://dsmr-reader.readthedocs.io/en/latest/installation.html

Er staat bij oudere versies ook bovenaan elke pagina een waarschuwing, maar ik kan me voorstellen dat die altijd even goed opvalt.

mvklingeren, 2021-02-07:
Ja, dat was het probleem. Thanks Dennis, kom er verder wel uit :))))

closed 2021-02-07 by mvklingeren

dennissiemensma, 2021-02-07:
Dank voor je snelle reactie. Ik denk dat ik uiteindelijk de oudere documentatie versies maar wat sneller ga uitfaseren. Je bent namelijk niet de eerste die hier tegenaan liep ;-)

mvklingeren, 2021-02-15:
Hi Dennis,

Ik heb inderdaad die melding over het hoofd gezien; via google simpelweg
gezocht op: installation dsmr.....
weet niet dat je er veel aan kan doen als mensen niet lezen ; -)

-

Had gisteren na het aansluiten van een lamp het probleem dat hier een
aardlekschakelaar eruit lag. Kan gebeuren...
maar liep tegen een heel vervelend probleem aan waardoor het pid lock
bestand van gunicorn in de weg bleef zitten. En dsmr badgateway bleef geven
(duurde even eer ik daar achter kwam /logs bleef ook leeg):

sudo rm /var/tmp/gunicorn--dsmr_webinterface.pid
sudo supervisorctl restart dsmr_webinterface

was de oplossing (kwam ik tegen uit jouw bericht voor andere gebruiker); en
snap wel wat het probleem is
maar tegen stale pid lock, valt daar niet iets aan te doen?

Als *'PS -A'* geen pid voor unicorn geeft; dan mag er toch ook geen .pid
file zijn? Dan kan je de .pid wel verwijderen..
maar de dsmr user mag niet op dat niveau erbij denk ik.. (tenzij deze wat
meer rechten krijgt)

Denk ik nu te simpel en is dit niet zo makkelijk op te lossen?

Groeten,
Milco

dennissiemensma, 2021-02-15:
Ik kan wel een keertje kijken of ik de PID files kan verplaatsen naar `/tmp/` ipv `/var/tmp/`. De laatstgenoemde wordt nu gebruikt en overleeft een reboot, dat is de oorzaak. Alleen weet ik niet meer of ik het ooit met goede reden juist in `/var/tmp/` had gezet.