dsmrreader/dsmr-reader #1278: Incorrect graph display when missing measurements

RoboMagus, 2021-02-08:
I've had a short circuit last night causing my Raspberry PI with DSMR-reader to go offline for a few hours.
When looking at the graphs around the outage time it simply continues unbroken, even though there's a gap of multiple hours (See image below. Outage occurred around 0.30, and I've powered back on at 6.00-ish) .
I would assume the graphs to correctly reflect missing data and have a time-continuous X-axis, rather than per sample basis.

[image]

( For those wondering about the high power consumption, I had to turn on my pond heater due to the intense cold. I didn't like the idea of my fish going belly-up ;) )

dennissiemensma, 2021-02-08:
Bedankt voor je melding! Ik zal hier in de toekomst een keer naar kijken.

Ik kan me vaag herinneren dat ik eerder een poging gedaan heb. Het probleem was dat alle datapunten geen vast interval hebben en het daarom gebaseerd wordt op de datapunten. Dat maakt het tevens wat minder makkelijk om om gaten in gegevens af te vangen. 

Wellicht dat het toch mogelijk is via de grafiekentool die ik gebruik, maar dat zal ik eerst moeten onderzoeken.

dennissiemensma, 2021-02-09:
Bedankt voor je aanvulling. Ik was daar al bang voor. Dat zeggende hebben zal ik je bevindingen tzt meenemen, want het klinkt sowieso al beter om gebruik te maken van de native datetime in JS ipv wat ik nu heb gedaan. 
Wellicht had ik er destijds een goede reden voor, maar dat kan ik me niet meer heugen. Mogelijk iets met vertalingen of toch die zoom waar je nu ook tegenaan loopt.

RoboMagus, 2022-05-13:
Any update on a fix for this issue?
If not, I'm willing to give it a go and create a PR for this ;)

dennissiemensma, 2022-05-13:
Bedankt voor je vraag. Dit staat momenteel gepland voor "een toekomstige release", maar kan ook nooit opgepakt worden.

Het lastige hieraan is dat het een foutsituatie is die niet vaak voor komt. En als die voorkomt, dan wordt het gat (na fixen inkomende data) eigenlijk alweer verdreven door nieuwe data. 
Mocht het zelfs regelmatig of terugkerend voorkomen bij iemand, dan zou ik diegene eerder vragen om de oorzaak van missende data op te lossen.

Daarnaast is DSMR-reader technisch zo opgezet, dat het (achteraf gezien) niet heel triviaal is om de manier van data tussen weergave/browser en server/database aan te passen, laat staan de structuur voor de grafieken. Normaal gesproken/tegenwoordig zou het gewoon via een API lopen en alle logica qua weergave in de browser. De huidige opzet is redelijk legacy, helemaal met de kennis van nu.

RoboMagus, 2022-05-14:
Het is gelukkig vrij goed op te lossen met bijna enkel het aanpassen van de echarts configuratie.
Ik zal binnenkort eens kijken of ik lokaal de aanpassingen kan maken waarmee ik verwacht dat dit probleem verholpen zal zijn.
Als dit lukt zal ik er tzt wel een PR voor openen.

dennissiemensma, 2023-10-23:
Ik heb hier een keer naar gekeken, maar ik kreeg het niet werkend. Het is helaas ook inherent aan hoe DSMR-reader de data ontsluit aan echarts. 

Mocht iemand een echarts-config werkend krijgen, waarbij de data uit DSMR-reader niet verandert (datetime formatting wijzigen is prima), dan wil ik het nog wel opnieuw bekijken tzt.

closed 2023-10-23 by dennissiemensma