dsmrreader/dsmr-reader #1279: Temperatuur API gaat fout

GieltjE, 2021-02-08:
Zoals we allemaal weten is het momenteel koud, echter denkt DSMR reader van niet:

[image]
[image]
[image]

```
DSMR-READER
    App / Python / Database                                                       v4.11 / v3.8.7 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             680733
    Consumption records electricity / gas (est.)                                               638684 / 38388

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_consumption_electricityconsumption                                                     542 MB
```

jeroenpeters1986, 2021-02-08:
Hoi @GieltjE,

DSMR Reader kijkt naar https://data.buienradar.nl/2.0/feed/json
Daar wordt de grondtemperatuur van gebruikt (zie ook: https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_weather/services.py#L50)

Als ik de feed bekijk voor de Bilt dan zie ik dit:
[image]

Dus wat mij betreft klopt het wel

GieltjE, 2021-02-08:
Inderdaad, is het misschien een optie om dan alle twee de temperaturen te gaan opslaan?
Ze lijken mij alle twee relevant namelijk.

jeroenpeters1986, 2021-02-08:
Ik denk niet dat dat gaat gebeuren, in andere threads heb ik gelezen dat Dennis (die DSMR Reader ontwikkelt) dit eigenlijk al buiten de scope van het project vond gaan. Ik snap dat jij vindt dat die relevant is, maar dan zou je ook nóg meer op kunnen slaan om je verbruik te duiden aan de hand van het weer en voor je het weer synchroniseer je Buienradar volledig.

Maar laat ik verder niet voor mijn beurt spreken, het woord is denk ik aan Dennis hierin, ik zal het issue ook bij hem zetten

dennissiemensma, 2021-02-08:
Bedankt voor jullie toelichting! Ik sluit me aan bij Jeroen qua opslag van temperaturen en houd dat zo simpel mogelijk. Het is, net zoals bijvoorbeeld alle prijsberekeningen in DSMR-reader, meer bedoeld als een indicatie om een gevoel te geven van wat er op een dag heeft gespeeld.
Dit omdat voor het verbruik veel meer factoren in meespelen, wat niet allemaal in DSMR-reader kan. Het beste voorbeeld zijn [graaddagen](https://www.mindergas.nl/degree_days/explanation) en daarvoor is wellicht Mindergas (of een andere tool) handiger om te gebruiken.

Overigens heb ik ooit de grondtemperatuur (10 cm hoogte) gekozen, omdat die destijds regelmatig iets lager was dan de andere temperatuur (op 150 cm hoogte). Ik zie nu overigens ontzettend wisselende metingen, waarbij beide hoogtes soms gelijk zijn of amper verschillen.

Het enige wat ik eventueel zou kunnen doen is om in een toekomstige release het **gemiddelde** van beide waarden pakken. Dan wordt het een gemiddelde rond 75 cm hoogte.

GieltjE, 2021-02-09:
Denk dat ze beiden relevant zijn, de ene is (om en nabij waarschijnlijk) de buitensteen temperatuur van je huis, de ander is de lucht die je naar binnen circuleerd.
Weet niet of het beter word van het middelen tussen deze twee.

Snap heel goed dat je alles simpel wilt houden, aan de andere kant is het voor de eind gebruiker weer frustrerend als ze voor het totaal plaatje (denk ook aan b.v. de watermeter) meerdere stukken software nodig hebben (terwijl deze al mooi statistieken e.d. kan laten zien).

jeroenpeters1986, 2021-02-09:
Ik snap wat je zegt, maar dan moet je als software ontwikkelaar dat ook allemaal (in je vrije tijd) bijhouden. Dat is wel een nadeel aan veel externe services, en die kunnen ook weer onderhevig aan veranderingen zijn. 

Watermeters laten zich op dit moment sowieso nog slecht uitlezen. Er zijn wel mensen die dat met camera’s en tekstgerkenning proberen maar daar is volgens mij sowieso nog geen betrouwbare oplossing voor in Nederland.

dennissiemensma, 2021-02-09:
@GieltjE ik begrijp dat het superhandig zou zijn voor gebruikers, alleen is DSMR-reader niet gemaakt (of bedoeld) om dat soort problemen op te lossen. Ik denk namelijk dat het een utopie is om 1 systeem te hebben wat alles kan. 

Helemaal met alle software en tools die er tegenwoordig zijn, en wat er allemaal nog aan komt, lijkt het me onmogelijk om alles met alles te integeren binnen een systeem wat alles snapt. En zelfs al zou er zo'n systeem zijn, dan heeft iedereen weer zijn eigen voorkeur of mening over hoe dingen weergegeven moet worden.
Om nog maar te zwijgen over eventueel verschillende merken/fabrikanten van databronnen (meters, sensoren, omvormers, etc) of, wat Jeroen terecht aankaart, diverse externe services.

Het doel van DSMR-reader is simpelweg het uitlezen van de slimme meter, daar een weergave van te maken. En om er voor te zorgen dat het de gegevens deelt met eventuele andere systemen.
En dat laatste punt zie ik eigenlijk als enige oplossing. DSMR-reader probeert daar twee standaarden voor te gebruiken voor het ontsluiten van de gegevens, namelijk MQTT (en enkele externe API's) voor het *pushen* van data naar buiten DSMR-reader. En als tweede een REST API voor het mogelijk maken om gegevens te *pullen* uit DSMR-reader. 

Daarmee is het aan de gebruiker of en hoe de gegevens vervolgens gebruikt worden. De een zet het in HA, de ander in een eigen dashboard en weer iemand anders gooit het naar InfluxDB, waar ook andere (eigen) data in staat. Er zijn zoveel smaken, waarbij ik overtuigd ben dat DSMR-reader daar nooit in kan faciliteren.

dennissiemensma, 2021-02-10:
@GieltjE zijn de temperaturen vandaag wat beter voor je? Mocht middelen van beide velden toch nog een verbetering zijn, laat het weten. En anders laat ik het huidig gebruikte veld voor wat het is.

closed 2021-02-11 by GieltjE