dsmrreader/dsmr-reader #1280: Alert versturen op basis van bijv. dagresultaat

mrwolf78, 2021-02-09:
Ik gebruik DSMR o.a. om de opbrengst van mijn zonnepanelen te monitoren. Nu is afgelopen week na een stroomstoring de omvormer defect geraakt, waardoor er dus geen energie meer wordt teruggeleverd.

Is het mogelijk om op basis van bepaalde dagwaarden (bijv. 0 watt teruggeleverd over de afgelopen dag) een alert te sturen via mail/Telegram etc.?

Ik krijg nu al wel dagelijks een Telegram-bericht met de samenvatting, maar die lees ik niet nauwkeurig genoeg om ineens een uitval van teruglevering op te merken. Dus een extra alert op basis van vooraf ingestelde waarden zou een mooie feature zijn.

dennissiemensma, 2021-02-09:
Bedankt voor je verzoek. Ik kan me goed voorstellen dat je zo'n feature voor je eigen situatie wil, maar ik denk dat het simpelweg niet in DSMR-reader past als standaardfunctionaliteit. 

Waarschijnlijk ben je het meeste geholpen door een script of plugin te maken wat dagelijks, via de API van DSMR-reader, kijkt naar de teruglevering van de vorige dag. En bij nul-waardes dat telegram-bericht stuurt. 

Wellicht ben je zelfs nog beter geholpen door te kijken naar PVOutput, gezien dat helemaal gemaakt is voor omvormers en teruglevering. Ik zie dat er een soortgelijke feature beschikbaar is daar als "Low Output Alert": https://pvoutput.org/help.html#alerts

dennissiemensma, 2021-02-10:
Mocht het niet lukken met PVOutput, laat het vooral weten. Via een script of plugin aan DSMR-reader kun je wellicht hetzelfde bereiken, al is dat dan dus puur gemaakt voor je eigen situatie.

closed 2021-02-10 by dennissiemensma

mrwolf78, 2021-02-10:
Ik was direct na jouw opmerking aan de slag gegaan met pvoutput, dank voor de tip! Echter, ik krijg geen alert voor elkaar. 
[image]

De omvormer is nog steeds stuk, er wordt 0 watt teruggeleverd. Maar er wordt geen alert getriggerd. Maar dit is misschien meer iets voor het pvoutput-forum ;)

dennissiemensma, 2021-02-10:
Dank voor je terugkoppeling. Je kunt het idd daar even vragen. 

Ik weet wel dat sommige features overigens alleen beschikbaar zijn daar via een donatie ($15 per jaar). Mogelijk valt dit daar ook onder, al zou ik daar dan een waarschuwing of melding van verwachten. In een andere sectie staat het wel impliciet namelijk als dono-feature: https://pvoutput.org/help.html#donations-feature-bonus

```
Output Performance Alerts

Send an alert when the live energy output of your system
falls below a percentage threshold against another system.
```

Maar vraag het dus gerust op het forum daar voor de zekerheid.