dsmrreader/dsmr-reader #1281: Solaredge API support

RezzZ, 2021-02-10:
Sinds kort zonnepanelen op m'n dak en net als de meerderheid heb ik een solaredge invertor.
Hoewel niet perse relevant voor een slimme meter, en dus dsmr reader, geeft een overzicht van de totaal opgewekte energie een completere plaat bij dagstatistieken ipv dat opgewekte energie die meteen verbruikt wordt niet afgeleid kan worden uit de dsmr statistieken. Nu zien we alleen het surplus aan teruggeleverde energie, niet het werkelijk aantal opgewekte kWhs.

Aangezien weersstatistieken ook geïntegreerd zijn kan dit misschien ook een geweldige toevoeging zijn aan dsmr reader en ik denk niet super moeilijk te realiseren.

Iets om wel rekening mee te houden is het maximale aantal apicalls per dag van 300 dus zou mooi zijn als dan ook meteen deze statistieken via mqtt gepubliceerd worden zodat bv een homeassistant niet ook de API hoeft te bevragen?

jeroenpeters1986, 2021-02-10:
Hallo @RezzZ deze vraag is al heel vaak voorbij gekomen hier. Het antwoord van Dennis (de ontwikkelaar) is altijd dezelfde geweest dus ik neem de vrijheid dit te herhalen. 

DSMR reader is alleen bedoeld om de slimme meter uit te lezen, dat is de scope. Er zijn een paar export mogelijkheden naar andere services en meerdere manieren om zelf de data eruit te halen om te combineren met andere diensten, bijvoorbeeld een dienst voor je zonnepanelen.

Een ander iets wat meespeelt is natuurlijk dat SolarEdge niet de enige leverancier van omvormers is en dat je zo alsnog mensen met een andere leverancier teleurstelt. 

Ik zal het issue dan ook alvast sluiten, maar misschien dat Dennis alsnog wel reageert.

closed 2021-02-10 by jeroenpeters1986

RezzZ, 2021-02-10:
Ai, en ik had nog wel gezocht op solaredge in de issues (open en closed) er waren er vier gesloten maar geen die zover ik zag over deze vraag gingen.

Het weer heeft in principe ook niets met een slimme meter te maken toch?

jeroenpeters1986, 2021-02-10:
Nee dat klopt, maar dat is natuurlijk geen reden om er andere dingen in te zetten ;-)

Ik dacht zelfs dat Dennis in een ander ticket al eens had gezegd dat hij dat achteraf ook niet had willen/moeten doen. Nu dat erin zit (als sinds zo’n beetje de initiële release) en mensen het gebruiken wordt het er niet uitgehaald, maar ook niet uitgebreid (daar was in de afgelopen week nog een issue over)

dennissiemensma, 2021-02-10:
@RezzZ ik denk dat je het meeste geholpen bent met PVOutput. DSMR-reader is vooral gericht op het uitlezen en visualiseren van de slimme meter. 
Zie vooral #9 en de referenties daar qua PVOutput. Ik denk namelijk dat PVOutput zich daar in specialiseert en een meer te bieden heeft op dat gebied dan DSMR-reader. Inhoudelijk kan ik je daar niet mee helpen, maar er zijn meerdere gebruikers die het (blijkbaar) toch goed in PVOutput hebben gekregen.

----

Voor wat achtergrondinfo: 
Zaken zoals de weerstatistieken zijn er vooral in gekomen omdat ik dit project destijds initieel voor mezelf had gemaakt en dat het pas later opgepikt werd door andere gebruikers. Tegen de tijd dat de meer exotische wensen kwamen (en nog steeds komen), was het te laat om dat alsnog te verwijderen zonder gebruikers te veel te benadelen. 

Dat zeggende hebben kan het best zijn dat ik in de toekomst een grote rework release maak, die erg versimpeld is en een hoop features dropt. Ik ben door de jaren heen overtuigd geraakt dat er geen systeem is wat alles gaat kunnen en voor iedereen werkt. Dan zorg ik er liever voor dat de dingen die dit project doet robuust en voorspelbaar zijn, zodat het makkelijk te koppelen is met andere systemen of dashboards. 
Want daar zijn namelijk genoeg goede projecten of tools voor en ik denk niet dat DSMR-reader nog een extra home-dashboard hoeft te zijn, ook al begrijp ik dat het situationeel heel handig is voor iemand. 
Dat is de reden waarom ik dit soort verzoeken zoveel mogelijk buiten DSMR-reader houd.