dsmrreader/dsmr-reader #1283: Foutmelding na energiecontract ingevuld te zijn

skank01, 2021-02-11:
Na volgende gegevens in energie contract in te vullen: (kan ik trouwens zonder einddatum werken, want die is er eigenlijk niet?)

[url=https://ibb.co/10HBXW8][img]https://i.ibb.co/2FXRM4g/energie.jpg[/img][/url]

Krijg ik volgende foutmelding:
Server Error
Sorry, something unexpected happened.
Exception:
ValueError: Cannot use None as a query value
Traceback:
  File "/usr/local/lib/python3.9/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 614, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/sites.py", line 233, in inner
    return view(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 1653, in add_view
    return self.changeform_view(request, None, form_url, extra_context)

  File "/usr/local/lib/python3.9/site-packages/django/utils/decorators.py", line 43, in _wrapper
    return bound_method(*args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 1534, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 1573, in _changeform_view
    form_validated = form.is_valid()

  File "/usr/local/lib/python3.9/site-packages/django/forms/forms.py", line 177, in is_valid
    return self.is_bound and not self.errors

  File "/usr/local/lib/python3.9/site-packages/django/forms/forms.py", line 172, in errors
    self.full_clean()

  File "/usr/local/lib/python3.9/site-packages/django/forms/forms.py", line 374, in full_clean
    self._clean_fields()

  File "/usr/local/lib/python3.9/site-packages/django/forms/forms.py", line 395, in _clean_fields
    value = getattr(self, 'clean_%s' % name)()

  File "/dsmr/dsmr_consumption/forms.py", line 14, in clean_electricity_delivered_1_price
    self._validate_price_defined_for_range('electricity_delivered_1_price')

  File "/dsmr/dsmr_consumption/forms.py", line 45, in _validate_price_defined_for_range
    existing_contracts = EnergySupplierPrice.objects.exclude(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 942, in filter
    return self._filter_or_exclude(False, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 962, in _filter_or_exclude
    clone._filter_or_exclude_inplace(negate, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 969, in _filter_or_exclude_inplace
    self._query.add_q(Q(*args, **kwargs))

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1358, in add_q
    clause, _ = self._add_q(q_object, self.used_aliases)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1377, in _add_q
    child_clause, needed_inner = self.build_filter(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1237, in build_filter
    return self._add_q(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1377, in _add_q
    child_clause, needed_inner = self.build_filter(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1237, in build_filter
    return self._add_q(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1377, in _add_q
    child_clause, needed_inner = self.build_filter(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1319, in build_filter
    condition = self.build_lookup(lookups, col, value)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/query.py", line 1170, in build_lookup
    raise ValueError("Cannot use None as a query value")

dennissiemensma, 2021-02-11:
Bedankt voor je melding! Ik zal het fixen in de volgende release.

dennissiemensma, 2021-02-11:
De contracten moeten datums hebbenzodat het bepalen van de prijzen voor een dag enigzins werkbaar is, al snap ik dat het wat praktischer voor je zou zijn als ze leeg zouden kunnen blijven.

skank01, 2021-02-11:
Als ik het zo lees, is de foutmelding door invullen foutieve datum?
Maar ik heb een datum ingevuld (niet leeggelaten) dus wat moet ik dan wel invullen?

dennissiemensma, 2021-02-11:
Zie bovenin je screenshot. Ik dacht dat je dit wilde melden als fout. Je vult 'Eneco' in als startdatum.

[image]

dennissiemensma, 2021-02-11:
In de volgende release zal ik zorgen voor een fatsoenlijke fout. Zoiets als dit:

[image]

skank01, 2021-02-11:
Das toeval ;) Nu lukt het idd, ik had het niet eens gezien dat ik de naam op de plaats van de datum gezet had :)
Toevallig dus de bug gemeld :)

skank01, 2021-02-11:
Ik krijg volgende kosten nadat ik bovenstaande aan gegevens in gevuld heb
[url=https://ibb.co/8bFFpZP][img]https://i.ibb.co/qCbbfQm/kosten.jpg[/img][/url]
Dat kan toch niet kloppen?
Ik zou dan vandaag 66 euro verbruikt hebben dag... 34 nacht (en die is nog niet om) en dan nog es 115 gas

dure dag. lol

Waar mis ik iets?

dennissiemensma, 2021-02-11:
De prijzen in in Euro's en niet in centen. Ik denk dat dat de oorzaak is. Probeer het eens met formaat `0,1903` ipv `19,03` .

skank01, 2021-02-11:
oh hoe stom van me, excuses thx !

dennissiemensma, 2021-02-11:
Top!

dennissiemensma, 2021-02-11:
Ik heb de foutafhandeling aangepast voor de volgende versie, nogmaals dank voor je melding.

closed 2021-02-11 by dennissiemensma

dennissiemensma, 2021-02-17:
Onderdeel van [v4.12](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.12.0)