dsmrreader/dsmr-reader #1285: Live verbruik blijft 0W in dashboard en verspringt soms

TheCrow-zz, 2021-02-11:
Als ik inlog in het DSMR Dashboard staat er bij Live consumed 0W en bij returned vespringt hij wel af en toe. 
Voorheen kon ik wel live mijn verbruik zien, het updaten van 4.8 naar 4.11 heeft ook niet geholpen.
Met sudo supervisorctl status staat alles op Running.

sudo supervisorctl restart all heeft helaas het probleem niet doen verhelpen, en volledige reboot van de Raspberry pi 4 ook niet.

Soms verpringd hij de waardes en laat hij waardes zijn bij Live consumed en stopt hij bij returned. 
En returned is dat voor de zonnepanelen?

dennissiemensma, 2021-02-11:
Bedankt voor je melding. Zou je een `dsmr_debuginfo` kunnen delen? Deze vind je op de About-pagina in je DSMR-reader-installatie.

En gebruik je Docker?

dennissiemensma, 2021-02-14:
@TheCrow-zz ben je hier nog verder mee gekomen?