dsmrreader/dsmr-reader #1287: How to enable Weather update?

vdbrink, 2021-02-11:
I see in the config : Backend (dsmr_backend) > Weather update which is not active.
How can I activate this?

I can't find anything about it in the documentation.

I see on github in the code an endpoint DSMRREADER_BUIENRADAR_API_URL Is this something which must be defined?

dennissiemensma, 2021-02-11:
Je kunt dit aanvinken in de configuratie, onderaan de tussenpagina van de screenshot hier:
https://dsmr-reader.readthedocs.io/en/latest/configuration.html

Die linkt direct door naar de admin-pagina waar je moet zijn. Daar kun je het aanvinken en een station kiezen.

vdbrink, 2021-02-11:
Bedankt, gevonden! onder het kopje 'Weergegevens-integratie' op de url /admin/dsmr_weather/weathersettings/1/change/

closed 2021-02-11 by vdbrink