dsmrreader/dsmr-reader #1292: Server error na upgrade

Kruijd, 2021-02-13:
Na de upgradepoging naar de laatste versie van DSMRReader V4 krijg ik de volgende foutmelding. Blijkbaar gaat het bij de database migratie mis.  
Bij voorbaat dank voor de hulp!
Jan

Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying auth.0012_alter_user_first_name_max_length...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.IntegrityError: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(170) already exists.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 85, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 245, in handle
    fake_initial=fake_initial,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 228, in apply_migration
    self.record_migration(migration)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 243, in record_migration
    self.recorder.record_applied(migration.app_label, migration.name)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/recorder.py", line 87, in record_applied
    self.migration_qs.create(app=app, name=name)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 447, in create
    obj.save(force_insert=True, using=self.db)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 754, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 792, in save_base
    force_update, using, update_fields,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 895, in _save_table
    results = self._do_insert(cls._base_manager, using, fields, returning_fields, raw)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 935, in _do_insert
    using=using, raw=raw,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1254, in _insert
    return query.get_compiler(using=using).execute_sql(returning_fields)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1397, in execute_sql
    cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "django_migrations_pkey"
DETAIL:  Key (id)=(170) already exists.

>>>>> [!] Executing database migrations failed again! <<<<<
[... 194 lines cut ...]
dsmr_stats
 [X] 0001_squashed_0016_drop_stats_settings
 [X] 0002_regenerate_missing_gas_data
 [X] 0003_hour_statistics_gas_default
 [X] 0004_hour_statistics_gas_default_retroactive
 [X] 0005_statistics_exportverbose_names
 [X] 0006_min_max_temperature_statistics
 [X] 0007_min_max_temperature_statistics_retroactive
 [X] 0008_meta_names
 [X] 0009_statistics_editable
 [X] 0010_statistics_models_index
 [X] 0011_note_model_index
 [X] 0012_electricity_statistics
 [X] 0013_all_time_low
 [X] 0014_day_total_cost_index
 [ ] 0015_fixed_daily_cost
dsmr_weather
 [X] 0001_weather_models
 [X] 0002_meta
 [X] 0003_next_sync_setting
 [X] 0004_next_sync_setting_retroactive
 [X] 0005_weather_refactor_scheduling
 [X] 0006_schedule_weather_update
sessions
 [X] 0001_initial

dennissiemensma, 2021-02-13:
Je zou deze kunnen proberen: https://github.com/dsmrreader/dsmr-reader/issues/866#issuecomment-583565825

En daarna opnieuw proberen `deploy.sh` uit te voeren.

Kruijd, 2021-02-13:
Dat heeft het gedaan, bedankt Dennis!
complimenten voor dsmr reader en je voortreffelijke service!

closed 2021-02-13 by Kruijd