dsmrreader/dsmr-reader #1293: Feature request: Totaal verbruik archive screen & doorlinken naar dag view

Koal4, 2021-02-14:
Is het misschien mogelijk om totaal verbruik ook te vermelden onder de 'mouse hoover' in archive screen?

[image]

Ik begrijp dat je dit ook aflezen vanuit de grafiek maar daar kan ik niet de onderscheid maken of het no 7 of 8 is.,

Tevens kwam ik ook meteen op het idee of het mogelijk is om op een dag (onder maand view) te kunnen kunnen klikken zodat je meteen naar de de daily view gaat?

dennissiemensma, 2021-02-14:
Bedankt voor je verzoek. Ik zal er ooit opnieuw naar kijken, maar ik weet wel dat het niet heel triviaal is om werkend te krijgen. Hier heeft ook iemand eerder naar gekeken.

dennissiemensma, 2021-09-22:
Ik heb hier nogmaals naar gekeken en het is inderdaad helaas niet heel triviaal. Dus ik laat dit voor nu voor wat het is.

closed 2021-09-22 by dennissiemensma