dsmrreader/dsmr-reader #1294: DSMR Reader (latest) won't start anymore

remyderuysscher, 2021-02-14:
When upgrading docker container from xirixiz/dsmr-reader-docker:latest-v4.9.0-amd64 to latest
```

Server Error
Sorry, something unexpected happened.
Exception:

ProgrammingError: column dsmr_datalogger_dataloggersettings.override_telegram_timestamp does not exist
LINE 1: ...atalogger_dataloggersettings"."restart_required", "dsmr_data...
```

dennissiemensma, 2021-02-14:
Bedankt voor je vraag. Het lijkt erop dat de DB-migraties niet uitgevoerd zijn. Ik zou verwachten dat deze bij het starten van de container telkens uitgevoerd worden. Je kunt dit handmatig uitvoeren binnen de DSMR container met:
```
./manage.py migrate
```

remyderuysscher, 2021-02-14:
Helaas dit gaat niet goed.

```
bash-5.0# ./manage.py migrate
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  Applying dsmr_consumption.0017_energy_supplier_price_refactoring...Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.SyntaxError: syntax error at or near "WITH ORDINALITY"
LINE 6:                     FROM unnest(c.conkey) WITH ORDINALITY co...
                                                  ^

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/dsmr/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.9/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/base.py", line 85, in wrapped
    res = handle_func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/commands/migrate.py", line 243, in handle
    post_migrate_state = executor.migrate(
  File "/usr/local/lib/python3.9/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/usr/local/lib/python3.9/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/usr/local/lib/python3.9/site-packages/django/db/migrations/executor.py", line 227, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/usr/local/lib/python3.9/site-packages/django/db/migrations/migration.py", line 124, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/usr/local/lib/python3.9/site-packages/django/db/migrations/operations/models.py", line 508, in database_forwards
    alter_together(
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/schema.py", line 380, in alter_unique_together
    self._delete_composed_index(model, fields, {'unique': True}, self.sql_delete_unique)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/schema.py", line 411, in _delete_composed_index
    constraint_names = self._constraint_names(
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/schema.py", line 1188, in _constraint_names
    constraints = self.connection.introspection.get_constraints(cursor, model._meta.db_table)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/postgresql/introspection.py", line 148, in get_constraints
    cursor.execute("""
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/usr/local/lib/python3.9/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.ProgrammingError: syntax error at or near "WITH ORDINALITY"
LINE 6:                     FROM unnest(c.conkey) WITH ORDINALITY co...

```

dennissiemensma, 2021-02-14:
De foutmelding zegt me helaas niets. 

Welke DB-image gebruik je? En heb je die ook ge-upgrade?

remyderuysscher, 2021-02-14:
Nee alleen de container. Ik zal anders een db dump/restore proberen nadat er een nieuwe db is aangemaakt.

dennissiemensma, 2021-02-14:
Welke DB-versie gebruik je nu? Houd er rekening mee dat er momenteel issues zijn met recente Alpine-versies: https://github.com/dsmrreader/dsmr-reader/issues/1282#issuecomment-778618994. Je kunt waarschijnlijk niet de laatste gebruiken.

remyderuysscher, 2021-02-14:
Ik gebruik de standaard pg database van Synology DSM 6, dus geen docker postgress.

remyderuysscher, 2021-02-14:
Als ik een backup maak, database drop en container weggooi en vervolgens lege database en nieuwe container aanmaak dan zou ik verwachten dat er een hele nieuwe database wordt gemaakt zelfs dat gaat fout?

```

Server Error
Sorry, something unexpected happened.
Exception:

ProgrammingError: relation "dsmr_frontend_frontendsettings" does not exist
LINE 1: ...end_frontendsettings"."always_require_login" FROM "dsmr_fron...
                                                             ^

Traceback:

  File "/usr/local/lib/python3.9/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/dsmr/dsmr_frontend/mixins.py", line 10, in dispatch
    if not FrontendSettings.get_solo().always_require_login:

  File "/usr/local/lib/python3.9/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/compiler.py", line 1156, in execute_sql
    cursor.execute(sql, params)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/usr/local/lib/python3.9/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
```

dennissiemensma, 2021-02-14:
Ik weet niet precies hoe dat in combinatie met een niet-Docker-DB werkt. Krijg je dezelfde foutmelding als eerder, wanneer je de zelf migraties uitvoert?

Ik zie niet zo 1-2-3 een reden waarom je nu foutmeldingen krijgt, wanneer het eerder wel bij je werkte.

dennissiemensma, 2021-02-14:
Je kunt ook nog kijken of er een deel van de migraties is uitgevoerd, of geen enkele, met:
```
./manage.py showmigrations
```

tomelfring, 2021-02-14:
Als ik het goed zie, wordt WITH ORDINALITY pas ondersteund sinds postgres 9.4. Je geeft eerder aan dat je 9.3 gebruikt.

dennissiemensma, 2021-02-14:
Het zou ook nog wel eens een Django dingetje kunnen zijn inderdaad. Ik dacht dat versie 9 nog ondersteund werd, maar blijkbaar heeft Django al de ondersteuning onder de 9.5 gedropped.

https://docs.djangoproject.com/en/3.1/ref/databases/#postgresql-notes

```
Django supports PostgreSQL 9.5 and higher. psycopg2 2.5.4 or higher is required, though the latest release is recommended.
```

Wellicht was dit al langere tijd zo, maar loop je er nu tegenaan. Het zou iig de eerdere foutmelding verklaren en ook waarom de migraties niet automatisch (of uberhaupt) uitgevoerd kunnen worden.

Zou je eens willen proberen je Postgres-versie te upgraden?

dennissiemensma, 2021-02-14:
En in de volgende Django 3.2 (nu 3.1) wordt het nog een minor hoger:

```
Django supports PostgreSQL 9.6 and higher. psycopg2 2.5.4 or higher is required, though the latest release is recommended.
```

Dus wellicht goed om versie 10 of hoger te gebruiken. Dan kun je weer een tijdje vooruit.

remyderuysscher, 2021-02-15:
Ok dank, Ik blijf dan voorlopig op xirixiz/dsmr-reader-docker:latest-v4.9.0-amd64 zitten, deze werkt nog wel goed met postgres 9.3
Zodra ik naar een recente postgres container ben overgestapt doe ik een nieuwe poging.

remyderuysscher, 2021-02-15:
Postgres version < 9.4 not supported anymore.

dennissiemensma, 2021-02-15:
Top, laat maar weten als je er alsnog niet uit komt. Voor de volledigheid, het is niet alleen < 9.4, maar ook < 9.6 zodra Django 3.2 uit komt.

En op dit moment wordt ook alleen nog 9.6+ ondersteund door Postgres zelf tot eind dit jaar: https://www.postgresql.org/support/versioning/

Dit is wat minder een issue voor simpele thuis-databases, maar je doet je er goed aan om bij een postgres-container meteen een paar major versies hoger te pakken. Dan blijft dit gedoe je een tijdje bespaard in de toekomst.