dsmrreader/dsmr-reader #1296: Feature request: uitgebreide trends analyse

benoitwatt, 2021-02-15:
Hi Dennis,

Bedankt voor de ontwikkeling van deze applicatie! Werkt vlotjes met een BE meter! :)

Vraagje: is het mogelijk om de trends analyse uit te breiden naar een periode verschillend van de periode van de laatste 4 weken? Mijn doelstelling is om mijn energieverbruik te analyseren over een langere periode en daarop mijn verbruik af te stemmen.

Voornamelijk de grafiek 'AVERAGE ELECTRICITY CONSUMED EACH HOUR (IN %) WITHIN THE PAST 4 WEEKS' is super interessant. Ik zou echter de data range willen aanpassen (e.g. over 1 jaar, of over 2 maanden (e.g. zomer vs winter), of een bepaalde maand) en eventueel ook het verschil willen zien tussen weekdagen en weekend.

Mogelijk kan ik dit beter doen in een andere applicatie op basis van DSMR-reader data ? Heb je hier suggesties?

Alvast bedankt!
Mvg,
Benoit

dennissiemensma, 2021-02-15:
Bedankt voor je verzoek. Ik denk dat dit wel redelijk makkelijk voor je te maken is. Gezien het over gegevens direct uit de slimme meter gaat lijkt het me sowieso een mooie uitbreiding.

Momenteel staat het [hardcoded op 4 weken](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_frontend/views/trends.py#L39), maar daar is wel wat mee te spelen denk ik. Al weet ik niet tot hoever ik kan gaan totdat het te zwaar wordt.
Het meeste werk zit denk ik in het uitzoeken hoe dit via een control te regelen is.

dennissiemensma, 2021-03-20:
@benoitwatt ik heb hier inmiddels een opzet voor werkend, samen met wat restyling van de pagina:

[image]

Deze gaat mee met de volgende release.

closed 2021-03-20 by dennissiemensma

benoitwatt, 2021-03-21:
Mooi Dennis! Ik kijk er al naar uit!! Bedankt! :)

dennissiemensma, 2021-03-23:
Uitgebracht in [v4.14](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.14.0)

dennissiemensma, 2021-03-25:
@benoitwatt ben je inmiddels in de gelegenheid geweest om het te testen?

En nog een ongerelateerd vraagje, draai je DSMR-reader in Docker of via de handmatige installatie?

benoitwatt, 2021-03-25:
@dennissiemensma jazeker! Loopt vlot in lijn met verwachtingen! :)

Ik draai een manuele installatie op een raspbian.

Nog een ongerelateerd vraagje: wat zijn volgens jou de mogelijkheden om productie door zonnepanelen te integreren? Ik besef dat dit wat buiten de scope van DSMR reader ligt maar de use cases lijken mij wel interessant:
- simuleren van een thuisbatterij (hoeveel energie verbruik ik op moment van productie, hoeveel overschot is er welke kan opgestaan worden in een thuisbatterij (capaciteit is een variabele), en wanneer neem ik af van de batterij en volstaat dit?
- Hoeveel energie verbruik ik werkelijk: productie zonnepannelen + verbruik (som van de 1s waarden?)
- ...

Lijkt mij wel niet evident omwille van:
- data afhankelijk van merk omvormer (SMA is bv gemakkelijk uit te lezen via HA maar geen idee over de rest)
- Parameters batterij (laad efficiëntie/verlies)
- data kan eventueel manueel in juiste formaat gezet worden en dan manueel opgeladen worden (wel omslachtig)

Grtn!
Benoit

dennissiemensma, 2021-03-25:
Ik denk dat je daarvoor het beste naar PVOutput kan kijken. Dat platform is echt gemaakt voor alles rondom zonnepanelen. Er zijn gebruikers die hun bruto opwekking (uit omvormer) en hun netto verbruik (uit DSMR-reader) daar naar uploaden en er dan een totaalplaatje van maken. Alleen heb ik daar dus zelf geen ervaring mee. Zie issue #9 en vooral alle gebruikerscomments voor wat achtergrond.

Verder kwam iemand laatst met iets over thuisbatterij evaluatie, maar ik weet niet precies of/hoe goed dat werkt, wat het doet, maar welicht de moeite waard, zie #1271. Gaat overigens wel over de Belgische situatie, die een ander salderingsmodel hebben ofgaan krijgen (dacht ik).

benoitwatt, 2021-03-26:
Thanks! Gaat weer een hele nieuwe wereld voor mij open! :)

e036, 2021-04-15:
Thx @dennissiemensma. Had het graag in DSMR reader gehad maar snap je focus.