dsmrreader/dsmr-reader #1297: Supervisor-processen verplaatsen van /var/tmp/ naar /tmp/

dennissiemensma, 2021-02-15:
Uitzoeken of de PID-files van de Supervisor-processen in `/tmp/` kunnen, ipv `/var/tmp/`. Dit zodat ze na een (onvoorziene) reboot automatisch verwijderd worden.

Deployscript moet dan ook aangepast worden.

GieltjE, 2021-03-03:
Is /run hier niet beter voor? Minder problemen met systemd i.c.m. b.v. apache reverse proxy, en deze word normaliter ook opnieuw opgebouwd.

dennissiemensma, 2021-03-03:
De run-mappen zijn idd beter, alleen weet ik niet of ik daar tegen permissies aan ga lopen omdat de processen niet als root draaien. Ik zal het tzt even proberen.

GieltjE, 2021-03-03:
Nadat ik wisselde naar systemd (alsin compleet init systeem), waren er juist issues met /tmp, /var mocht apache daarintegen wel bij.
Wel even het volgende toevoegen aan het webinterface .service bestand zodat systemd automatisch /run/dsmr aanmaakt:
```
[System]
RuntimeDirectory=dsmr
```

dennissiemensma, 2021-03-03:
Bedoel je dat je uberhaupt geen Supervisor gebruikt? Want dan ga je denk ik ook geen last hebben van deze wijziging. Het gaat vooral om de PID-file van Gunicorn namelijk, die in de huidige versie een reboot overleeft.

GieltjE, 2021-03-03:
Inderdaad helemaal geen supervisor(d), was ik vergeten te vermelden.

dennissiemensma, 2021-11-22:
Hier nog wat achtergrond: https://wiki.debian.org/ReleaseGoals/RunDirectory

`/run` is een tmpfs, wat perfect is. Alleen zit ik dan nog met de permissies en dat lijkt wat lastiger, omdat die bij elke reboot hersteld moeten worden. Dus ik denk dat het uiteindelijk alsnog opkomt op iets in `/tmp/`

closed 2021-11-22 by dennissiemensma