dsmrreader/dsmr-reader #1298: Gas en temperatuur statistieken updaten niet live

reboot1983, 2021-02-16:
In het tabblad live grafieken wordt netjes iedere minuut een update gegeven van de 4 onderdelen van het stroomverbruik. Dit gaat per minuut en functioneert zoals verwacht. Echter, het gasverbruik per 5 minuten wordt niet 'live' (iedere 5 minuten) bijgewerkt. Evenals ieder heel uur ook geen update van de actuele temperatuur wordt getoond.

Als ik deze pagina bijvoorbeeld 5 minuten voor het hele uur open, had ik rond het hele uur een update van zowel gas als de actuele temperatuur verwacht. Een pagina-refresh is noodzakelijk om 'live' gegevens van deze twee onderdelen te krijgen.

dsmrreader v4.11.0

dennissiemensma, 2021-02-16:
Bedankt voor je melding. De pagina [ververst momenteel in z'n geheel na 15 minuten](https://github.com/dsmrreader/dsmr-reader/blob/v4.11.0/dsmr_frontend/templates/dsmr_frontend/live-graphs.html#L264). Ik kan die wel wat lager zetten op bijvoorbeeld 5 minuten, totdat dat iemand in de weg zit.

De refreshes zullen overigens altijd gebaseerd zijn op basis van je schermduur en niet de (systeem)tijd. Er is geen garantie dat een meter exact op het uur een nieuwe waarde heeft. 
Wellicht dat ik het gasverbruik ook ooit dynamisch laat polsen, net zoals de elektra-grafieken, maar dat is redelijk wat werk. Vandaar de huidige pagina refresh als workaround.

dennissiemensma, 2021-02-16:
Aangepast voor volgende release.

closed 2021-02-16 by dennissiemensma

dennissiemensma, 2021-02-17:
Onderdeel van [v4.12](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.12.0)

reboot1983, 2021-02-17:
Cool. Dankjewel voor de moeite.