dsmrreader/dsmr-reader #1299: 4.12 does not work on ARM systems

davidvf, 2021-02-18:
When starting the new DSMR container, i got the following output in the logs

```
[pi@hawk:~ $] docker logs -f dsmr
[ INFO ] Removing existing PID files...
[ INFO ] Creating log directory...
[ INFO ] Fixing /dev/ttyUSB* security...
[ INFO ] Verifying if the DSMR web credential variables have been set...
[ INFO ] Verifying database connectivity to host: owl.home with port: 5432...
[ INFO ] Database connectivity successfully verified!
[ INFO ] Running post configuration...
Fatal Python error: init_interp_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted

Current thread 0x76fb1390 (most recent call first):
<no Python frame>
Fatal Python error: init_interp_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted

Current thread 0x76fde390 (most recent call first):
<no Python frame>
Fatal Python error: init_interp_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted

Current thread 0x76f26390 (most recent call first):
<no Python frame>
[ INFO ] Checking for NGINX SSL configuration...
[ INFO ] ENABLE_NGINX_SSL is disabled, nothing to see here. Continuing...
[ INFO ] Checking for HTTP AUTHENTICATION configuration...
[ INFO ] ENABLE_HTTP_AUTH is disabled, nothing to see here. Continuing...
[ INFO ] Configuring DSMR in standlone datalogger mode....
[ INFO ] Starting supervisord...
[ INFO ] Logfiles can be found at: /var/log/supervisor/*.log and /tmp/supervisord.log
Fatal Python error: pyinit_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted

Current thread 0x76f76390 (most recent call first):
<no Python frame>
```

OS is fully up to date, running on 
```
[pi@hawk:~ $] cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

I found a similar issue on this website:
[https://github.com/linuxserver/docker-papermerge/issues/4](url)

Resolution:
```
wget http://ftp.us.debian.org/debian/pool/main/libs/libseccomp/libseccomp2_2.5.1-1_armhf.deb
sudo dpkg -i libseccomp2_2.5.1-1_armhf.deb
```

Retry container startup - all works fine now:
```
[pi@hawk:~ $] docker logs -f dsmr
[ INFO ] Removing existing PID files...
[ INFO ] Creating log directory...
[ INFO ] Fixing /dev/ttyUSB* security...
[ INFO ] Verifying if the DSMR web credential variables have been set...
[ INFO ] Verifying database connectivity to host: owl.home with port: 5432...
[ INFO ] Database connectivity successfully verified!
[ INFO ] Running post configuration...
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  No migrations to apply.

561 static files copied to '/var/www/dsmrreader/static'.
Updating password of superuser "admin"
[ INFO ] Checking for NGINX SSL configuration...
[ INFO ] ENABLE_NGINX_SSL is disabled, nothing to see here. Continuing...
[ INFO ] Checking for HTTP AUTHENTICATION configuration...
[ INFO ] ENABLE_HTTP_AUTH is disabled, nothing to see here. Continuing...
[ INFO ] Configuring DSMR in standlone datalogger mode....
[ INFO ] Starting supervisord...
[ INFO ] Logfiles can be found at: /var/log/supervisor/*.log and /tmp/supervisord.log
2021-02-18 11:48:36,220 INFO Included extra file "/etc/supervisor.d/supervisord.ini" during parsing
2021-02-18 11:48:36,220 INFO Set uid to user 0 succeeded
2021-02-18 11:48:36,264 INFO RPC interface 'supervisor' initialized
2021-02-18 11:48:36,277 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2021-02-18 11:48:36,279 INFO supervisord started with pid 16
2021-02-18 11:48:37,290 INFO spawned: 'dsmr_backend' with pid 18
2021-02-18 11:48:37,304 INFO spawned: 'dsmr_datalogger' with pid 19
2021-02-18 11:48:37,322 INFO spawned: 'dsmr_webinterface' with pid 20
2021-02-18 11:48:37,336 INFO spawned: 'nginx' with pid 21
2021-02-18 11:48:38,343 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2021-02-18 11:48:38,344 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2021-02-18 11:48:38,347 INFO success: nginx entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2021-02-18 11:48:40,696 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
```

dennissiemensma, 2021-02-18:
Thanks for reporting. I think you may want to report this to the docker project instead of it's not code related: https://github.com/xirixiz/dsmr-reader-docker/issues

closed 2021-02-19 by dennissiemensma

RoboMagus, 2021-02-23:
Had the same issue. Repored the bug on the dsmr-reader-docker project