dsmrreader/dsmr-reader #1301: Meterstanden in archive?

dataegg, 2021-02-18:
[image]

Zie attached screenshots. Is het mogelijk dat bij de weergave van het archief ook de op dat moment bekende meterstanden worden getoond?

dennissiemensma, 2021-02-18:
Ik denk dat zoiets wel mogelijk is, maar het wordt dan wel een benadering van de meterstanden. Ook om het met terugwerkende kracht te doen, als die telegrammen nog opgeslagen zijn (vandaar dat retentie ze niet allemaal verwijdert). De huidige statistieken houden namelijk alleen een diff bij en niet de absolute standen.

dataegg, 2021-02-18:
> Ik denk dat zoiets wel mogelijk is, maar het wordt dan wel een benadering van de meterstanden. Ook om het met terugwerkende kracht te doen, als die telegrammen nog opgeslagen zijn (vandaar dat [...]

Kijk maar eens even. Waar ik het nu voor nodig had heb ik het ook even voor teruggerekend. Ik denk dat meterstanden goed is voor het beeld. Meterstanden zijn 'normaal' als je b.v. excel gebruikt om zaken bij de houden juist de informatie die je hebt en al het andere wordt daar dan van afgeleid. DSMR reader doet dat natuurlijk anders en dat is ook logisch maar voor de beeldvorming en compleetheid zijn meterstanden denk ik een goede toevoeging.

dennissiemensma, 2021-02-24:
Ik zal deze eveneens toevoegen in het kader van datahistorie, omdat de meterstanden nu "verloren" gaan als iemand bijvoorbeeld alleen een backup heeft van de dagtotalen (wat altijd mijn advies is, want dat is "het belangrijkste" aan minimale gegevens door de jaren heen).

Nadeel is wel dat alle toekomstige backup's voor dagtotalen hierdoor wellicht twee keer zo groot worden, maar die "prijs" lijkt het me zeker waard.

dennissiemensma, 2021-02-24:
Verder zal ik het visualiseren ervan beperkt houden, omdat het anders teveel gegevens op 1 scherm worden en ik denk dat de diffs (wat nu getoond wordt) belangrijker zijn dan de absolute standen. 

Ik zal de meterstanden wel zichtbaar maken, maar dan in de admin, als naslagwerk.

dennissiemensma, 2021-02-24:
Gelukt. Datamigratie duurt wel een paar minuten op een Pi 4, afhankelijk van het aantal dagen + aantal metingen:

```
...
Data migration: Adding reading values to day statistics retroactively: 2021-02-17 (1879/1885)
Data migration: Adding reading values to day statistics retroactively: 2021-02-18 (1880/1885)
Data migration: Adding reading values to day statistics retroactively: 2021-02-19 (1881/1885)
Data migration: Adding reading values to day statistics retroactively: 2021-02-20 (1882/1885)
Data migration: Adding reading values to day statistics retroactively: 2021-02-21 (1883/1885)
Data migration: Adding reading values to day statistics retroactively: 2021-02-22 (1884/1885)
Data migration: Adding reading values to day statistics retroactively: 2021-02-23 (1885/1885)
```

Ontsluiten van de meterstanden dus alleen via de admin:

[image]

dennissiemensma, 2021-02-24:
Zoals verwacht is de partial backup nu wel stukken groter geworden. Bij mij zelfs een factor 10. Dat kan gevolgen hebben voor de opslag, alleen vind ik het lastig om te bepalen in hoeverre DSMR-reader dat moet managen. Zowel lokaal als evt via Dropbox. 

Ik kan de interval van de partials wel verhogen, zodat die bijvoorbeeld elke week een niet-roteerbare backup maakt ipv elke dag. Het is immers vooral om te voorkomen dat corrupte backups elke overschrijven en dat iemand slechts een week dataverlies heeft, tov een of meerdere maanden dataverlies (als er al een goede backup over was).

dennissiemensma, 2021-02-24:
Doorgevoerd. Draait straks alleen elke maandag om de schijfruimte wat te compenseren.  Dit is het denk ik absoluut waard, want in de plaats daarvan bevatten partials straks dus ook de meterstanden, waar het eerder alleen het verschil (verbruik) per dag bevatte. 
Plus dat het wellicht leuk was dat er een last-resort partial backup was van elke dag, maar dat de effectiviteit nog steeds afhangt van of die partials uberhaupt nog van de SD-kaart af te krijgen zijn (of in Dropbox staan).

De volledige backups draaien gewoon elke dag, maar overschrijven elkaar dus (nog steeds) na een week. Dat blijft ongewijzigd.

closed 2021-02-24 by dennissiemensma

dataegg, 2021-02-25:
Cool, ik wordt hier wel blij van. Dank voor je effort.

dennissiemensma, 2021-03-09:
Zojuist uitgebracht in [v4.13](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.13.0).