dsmrreader/dsmr-reader #1302: meterstanden wel in DB niet in GUI

remb0, 2021-02-21:
ik draai de laatste versie (en dan vanuit home assistant als addon) dit heeft het voordeel dat ik alles op 1 plek start en controleer. werkt fantastisch.
maar omdat ik vanaf 2019 (toen kreeg ik zonnepanelen) pas met DSMRReader ben gestart had ik mijn historie in Domoticz.   Dit heb ik middels de scripts 

geïmporteerd en deze zie ik ook netjes in de database terug. (wat jammer is dat ik het meerdere malen heb geprobeerd dus ik 2 keer de inhoud van 2014 t/m 2019-01 in de tables terug zie.

in de bijlage de twee tabels (heb de velden gekleurd om te laten zien.
dsmr_consumption_electricityconsumption
dsmr_consumption_gasconsumption
in de bijlage te zien.
[meterstanden.xlsx](https://github.com/dsmrreader/dsmr-reader/files/6017332/meterstanden.xlsx)

Nu vraag ik mij af wat ik het beste kan doen om mijn archief weer kloppend te krijgen in de interface.  
is er een schonings commando?
moet ik in de tables handmatig aan de slag (en hoe dan aangezien de PK's waarschijnlijk relaties hebben enzo (ben hier niet handig mee)

door de addon pgadmin ben ik wel in staat eenvoudig SQL te draaien of de inhoud te bekijken. alleen weet ik niet wat ik moet scripten zodat DSMR-reader het juiste kan [tonen.](url)

is er iemand die me verder op weg kan helpen?

dennissiemensma, 2021-02-21:
Dit komt omdat DSMR-reader bij het verwerken van gegevens altijd vooruit kijkt. Bij het maken van de dagtotalen pakt die telkens de vorige dag om te voorkomen dat die telkens weer door alle gegevens heen moet, wat traag is maar wat je juist in deze situatie eenmalig zou willen.
Ik kan wel een keertje iets maken wat het hergenereren forceert, specifiek voor de dagen die er nog niet zijn. Maar dat zal even duren voordat dat er is. 

Als workaround kun je **alle uur- en dagtotalen verwijderen** en dan zou DSMR-reader het automatisch met terugwerkende kracht moeten genereren, vanaf 2014. Dat zeggende hebben werkt dat wel het beste als je nog zoveel mogelijk (bron) gegevens hebt van de periode vanaf 2019. 
Want dat is het nadeel: ook die periode wordt dan opnieuw berekend en daar kunnen kleine verschillen in ontstaan als de brongegevens bijvoorbeeld inmiddels deels opgeschoond zijn.

Als je het laatste probeert
- **zorg dan sowieso voor een goede DB-backup voor de zekerheid**
- maak voordat je verwijdert een export van de dagtotalen en doe dat opnieuw daarna en controleer ook even willekeurig wat dagen van 2019+

Als het om wat voor reden niet lekker gaat, of als je te grote verschillen ziet met eerder, dan kun je iig terug met je backup.

remb0, 2021-02-22:
Bedankt voor je werk en hulp, wordt enorm gewaardeerd.
Backup is helder!  goed plan!
bedoel je de tabel:  public.dsmr_stats_daystatistics ?

als je in mijn bijlage kijk zie je dat ik een periode dubbel heb bijgeladen, moet ik die nog schonen op 1 of andere manier?
in dsmr_consumption_electricityconsumption en dsmr_consumption_gasconsumption

dennissiemensma, 2021-02-22:
Ja beide tabellen `dsmr_stats_daystatistics` en `dsmr_stats_hourstatistics`. Je hoeft die dubbele niet op te schonen, probeer het eerst maar op deze manier.

remb0, 2021-02-22:
dank.
Ik ben er gelijk mee aan de slag gegaan.
volgens mij gaat het niet goed met enkel een truncate van de tables en dan het draaien van de processen.
`
bash-5.1# tail -f /var/log/supervisor/dsmr_backend*
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
could not find a "pg_dump" to execute
pg_dump: warning: there are circular foreign-key constraints on this table:
pg_dump:   hypertable
pg_dump: You might not be able to restore the dump without using --disable-triggers or temporarily dropping the constraints.
pg_dump: Consider using a full dump instead of a --data-only dump to avoid this problem.
pg_dump: warning: there are circular foreign-key constraints on this table:
pg_dump:   chunk
pg_dump: You might not be able to restore the dump without using --disable-triggers or temporarily dropping the constraints.
pg_dump: Consider using a full dump instead of a --data-only dump to avoid this problem.
`

dennissiemensma, 2021-02-22:
Ik mis wat context, welke commands heb je uitgevoerd? Ik zou alleen foutmeldingen van de backend-log verwachten als dat gerelateerd is met het maken van een backup door DSMR-reader zelf.

remb0, 2021-02-22:
ik heb truncate table dsmr_stats_daystatistics en dsmr_stats_hourstatistics  gedaan.
tables leeg.
daarna de processen via backend gestart:
- Generate consumption data
- Generate day and hour statistics

ik zie in de tables hier niet bij veranderen. (weet ook niet of dat hoort)
en ik dacht ik maak nog een backup.  (die werd netjes gemaakt, maar gaf wel de logging toen ik keek).

remb0, 2021-02-22:
dus ik ben nu in dubio  of ik backup moet herstellen en dan  een half uur kwijt ben     of dat het automatisch goed komt ;)

dennissiemensma, 2021-02-22:
Je kunt het logging level even [op `DEBUG` zetten](https://dsmr-reader.readthedocs.io/en/v4/faq.html#logging-levels), zodat je in de backend log kunt zien wat die doet.

En dan gaat het vooral om `Generate day and hour statistics`. Ik weet alleen niet of die telkens 1 dag doet of meerdere. Hoe dan ook zou ik wel iets verwachten. Of een record in de dagtotalen of een fout in de log met waar die op wacht.

dennissiemensma, 2021-02-22:
Je kunt de back-up op elk moment terugzetten. Als je het te veel risico vindt, kun je het laten voor wat het is en wachten totdat ik er iets degelijks voor kan maken in een toekomstige release.

remb0, 2021-02-22:
ik ben bang dat ik even geduldig moet wachten.
mijn kennis reikt helaas nog niet zo ver om goed grip te krijgen op wat ik doe.

ook is het niet super belangrijk (is meer dat het supergaaf is voor de analyses :)

don't worry, geen druk/haast.

dennissiemensma, 2021-03-07:
Ik heb hier mogelijk wat voor kunnen maken. In de volgende v4.13 release kun je dit proberen:

```
sudo su - dsmr
./manage.py dsmr_stats_reconstruct_missing_day_statistics
```

closed 2021-03-07 by dennissiemensma

dennissiemensma, 2021-03-09:
Zojuist uitgebracht in [v4.13](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.13.0).

Laat maar weten of bovenstaande het voor je oplost. Maak sowieso even een backup vooraf (= altijd een goed idee bij elke vorm van upgraden of datamigratie).

remb0, 2021-03-09:
Super.  Ga ik zo snel mogelijk starten. Ik draai nu de addon in home assistant en daar is de update nog niet beschikbaar.  Zodra dat ik teste koppel ik terug.

remb0, 2021-03-14:
@dennissiemensma   super bedankt het werkt als een trein!   Eindelijk heb ik alles uit verschillende bronnen weer in 1 systeem.
niet dat ik het dagelijks bekijk, maar het is gewoon tof!   bedankt voor alle inspanningen. 

Mijn gas is ook goed gegaan.

dennissiemensma, 2021-03-14:
Fijn om te horen, dank voor de terugkoppeling!