dsmrreader/dsmr-reader #1303: Upgrade naar v4 gaat niet

eterpstra, 2021-02-21:
Hoi Dennis,

Voor weer een andere vriend probeer ik v3 naar v4 te upgraden.

ik loop tegen de volgende melding aan:

```
 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ImportError: No module named django.utils.version

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)
```

Ook zie ik dit als ik wil inloggen als DSMR user:

```
root@odroid:~# sudo su - dsmr
-su: /home/dsmr/.virtualenvs/dsmrreader/bin/activate: No such file or directory

Ik had verder geen foutmeldingen tijdens de upgraden dus weet even niet waar het fout is gegaan.

Systeem is volledige geupdate.
```

dennissiemensma, 2021-02-21:
Welke stappen van de upgrade heb je uitgevoerd? Het lijkt er op dat de virtualenv mist.

eterpstra, 2021-02-21:
Ik heb deze stappen gevolgd:

https://dsmr-reader.readthedocs.io/nl/v4/faq/v4_upgrade.html

dennissiemensma, 2021-02-21:
Wat staat er in de map:
```
/home/dsmr/.virtualenvs/
```

En gaf deze dan ook geen fouten halverwege?
```
sudo su - dsmr
./manage.py check
```

eterpstra, 2021-02-21:
```
root@odroid:/home/dsmr/.virtualenvs# ls
dsmrreader  v3-dsmrreader  v4-dsmrreader
```
```
dsmr@odroid:~/dsmr-reader$ ./manage.py check
System check identified no issues (0 silenced).
dsmr@odroid:~/dsmr-reader$
```

dennissiemensma, 2021-02-21:
Oke dat is vreemd. Je kunt nog even proberen of de rechten goed staan:

```
sudo chown -R dsmr:dsmr /home/dsmr/
```

En dan
```
sudo su - dsmr
which python3

./deploy.sh
```

Geeft dat nog steeds een fout?

eterpstra, 2021-02-22:
Bij deployen krijg ik hetzelfde te zien:

```
 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "dsmrreader/__init__.py", line 1, in <module>
    from django.utils.version import get_version
ImportError: No module named django.utils.version

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED
```

dennissiemensma, 2021-02-22:
Wat is de output van het which command als dsmr-user?

eterpstra, 2021-02-22:
```
dsmr@odroid:~/dsmr-reader$ which python3
/usr/bin/python3

```

dennissiemensma, 2021-02-22:
Kan het zijn dat `sudo su - dsmr` **geen** `source ~/.virtualenvs/dsmrreader/bin/activate` doet? 
Dat command zou in ieder geval in `/home/dsmr/.bashrc` moeten staan.

Los daarvan is het de vraag of dat command uberhaupt werkt, aan een van je eerdere foutmeldingen te zien. 

Bestaat deze wel?
```
/home/dsmr/.virtualenvs/dsmrreader/bin/activate
```

dennissiemensma, 2021-02-22:
En als het goed werkt, zou dat `which` command iets als `/home/dsmr/.virtualenvs/dsmrreader/bin/python3` moeten teruggeven.

jeroenpeters1986, 2021-02-23:
Hoi @eterpstra 
Als ik dit zo lees is het goed om de VirtualEnv nog eens aan te maken. Kan het zijn dat je hem eerder misschien draaide op `v3-dsmrreader`?
Hoe dan ook, met het volgende commando zet je de `dsmrreader` VirtualEnv even opzij (je kunt hem ook verwijderen overigens):
```
cd ~/.virtualenvs
mv dsmrreader broken-dsmrreader
```

Daarna kun je https://dsmr-reader.readthedocs.io/nl/v4/installation.html#virtualenv volgen om een nieuwe VirtualEnv te maken, misschien kun je daarna goed verder?

jeroenpeters1986, 2021-02-23:
Als dat ook niet werkt is misschien nog wel even raadzaam om te checken of je Python versie actueel is. Als ik het goed lees in de docs is minimaal 3.6 vereist.
Dat kun je checken met dit commando:
```
sudo su - dsmr
python3 --version
```

eterpstra, 2021-02-23:
Dat is inderdaad gelukt @jeroenpeters1986  

Echter ziet het er nu naar uit dat de webinterface supervisor ontbreekt of niet meer werkt:

```
dsmr_backend                     STOPPED   Not started
dsmr_datalogger                  RUNNING   pid 557, uptime 1121 days, 17:23:43
dsmr_webinterface                FATAL     can't find command '/home/dsmr/.virtualenvs/dsmrreader/bin/gunic
```

jeroenpeters1986, 2021-02-23:
@eterpstra dat is wel gek, dan draaide er toch iets. Maar die kun je als het goed is gewoon herstarten met `sudo supervisorctl restart dsmr_webinterface`.
Mocht dat niet werken, dan moet je misschien eerst nog even de pip requirements weer installerenmet:
```
sudo su - dsmr
pip3 install -r dsmrreader/provisioning/requirements/base.txt
```

eterpstra, 2021-02-23:
Helaas is het allemaal niet gelukt en daarom wil ik alles opnieuw gaan proberen. Ik loop nu tegen het volgende aan bij commando :

`root@odroid:~# sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt`

Krijg ik deze waslijst aan errros:

```
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site                                       -packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpcbz82f14
         cwd: /tmp/pip-install-hjrkewo0/pyyaml_b718d7a669104566aa2a0a01f812ee09
    Complete output (39 lines):
    running dist_info
    creating /tmp/pip-modern-metadata-z7oo42cy/PyYAML.egg-info
    writing /tmp/pip-modern-metadata-z7oo42cy/PyYAML.egg-info/PKG-INFO
    writing dependency_links to /tmp/pip-modern-metadata-z7oo42cy/PyYAML.egg-info/dependency_links.txt
    writing top-level names to /tmp/pip-modern-metadata-z7oo42cy/PyYAML.egg-info/top_level.txt
    writing manifest file '/tmp/pip-modern-metadata-z7oo42cy/PyYAML.egg-info/SOURCES.txt'
    skipping 'yaml/_yaml.c' Cython extension (up-to-date)
    reading manifest file '/tmp/pip-modern-metadata-z7oo42cy/PyYAML.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file '/tmp/pip-modern-metadata-z7oo42cy/PyYAML.egg-info/SOURCES.txt'
    creating '/tmp/pip-modern-metadata-z7oo42cy/PyYAML.dist-info'
    Traceback (most recent call last):
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", lin                                       e 280, in <module>
        main()
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", lin                                       e 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", lin                                       e 133, in prepare_metadata_for_build_wheel
        return hook(metadata_directory, config_settings)
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 113, in prepare_metadata_for_build_whee                                       l
        _run_setup()
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 72, in _run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 295, in <module>
        python_requires='>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*',
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/dist_info.py", line 36, in run
        bdist_wheel.egg2dist(egg_info.egg_info, dist_info_dir)
      File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 443, in egg2dist
        shutil.copy(license, os.path.join(self.distinfo_dir, license_filename))
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/posixpath.py", line 80, in join
        a = os.fspath(a)
    TypeError: expected str, bytes or os.PathLike object, not NoneType
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af                                       8d4ab1ed687d/PyYAML-5.4.1.tar.gz#sha256=607774cbba28732bfa802b54baa7484215f530991055bb562efbed5b2f20a45e (from htt                                       ps://pypi.org/simple/pyyaml/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*). Comma                                       nd errored out with exit status 1: /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrread                                       er/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpcbz82f14                                        Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pyyaml==5.4.1
ERROR: No matching distribution found for pyyaml==5.4.1
```

dennissiemensma, 2021-02-23:
Welke versie van `pip` en/of `pip3` gebruik je?

```
sudo su - dsmr
pip --version
pip3 --version
```

dennissiemensma, 2021-02-23:
Je kunt iig daar deze nog proberen:
```
pip install pip --upgrade
```

En als dat lukt (naar versie 20.x of 21.x oid) dan nogmaals de requirements install proberen.

eterpstra, 2021-02-23:
```
(dsmrreader) dsmr@odroid:~/dsmr-reader$ pip3 --version
pip 21.0.1 from /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip (python 3.6)

(dsmrreader) dsmr@odroid:~/dsmr-reader$ pip install pip --upgrade
Requirement already satisfied: pip in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages (21.0.1)
```

eterpstra, 2021-02-23:
ik denk dat ik maar de sd kaart ga formateren en opnieuw want volgens mij heb ik nu zoveel gedaan dat ik niet meer weet wat haha

dennissiemensma, 2021-02-23:
Je zou deze nog kunnen proberen: https://dsmr-reader.readthedocs.io/en/v4/faq.html#how-to-uninstall
En dan **niet de database weggooien** of de systeem packages deinstalleren. En dan weer een normale install, met de database nog actief.

Mocht dat niet werken en je RaspberryOS er weer opnieuw op gaat zetten, pakt dan meteen een recentere, zodat je daar wat langer op kan draaien komende jaren.

dennissiemensma, 2021-03-06:
@eterpstra is dit nog gelukt?

closed 2021-03-07 by dennissiemensma