dsmrreader/dsmr-reader #1304: Meer als 5 getallen achter de komma bij energie contracten

Diondk, 2021-02-22:
Hoi,

Mijn energie leverancier geeft mijn prijzen aan met 6 getallen achter de komma, echter kan ik deze niet toevoegen omdat er een maximum van 5 is. Uiteraard heb ik nu het laatste getal weggelaten, maar na verloop van tijd zit er dan een verschil in de berekening van de totale prijs, is het misschien mogelijk om dit veld aan te passen?

Tevens zou het misschien mogelijk zijn om achter Tarief 1 en 2 bij de energie contract toevoegen Hoog een laag tarief weer te geven? 

Bij het toevoegen van je energie contract is het nu een beetje onduidelijk welk tarief je voor wat moet hebben. 

Groet,
Dion

dennissiemensma, 2021-02-22:
Bedankt voor je vraag. Ik zal in een toekomstige release de nauwkeurigheid vergroten.

Wat betreft tarief 1/2, de reden dat het er niet bij staat is omdat het in Nederland verschilt met Belgie. En de tariefnamen die je zelf kunt instellen kan ik daar niet gemakkelijk tonen. Ik kan er wel wat extra uitleg bij zetten tzt. 

Als je in NL woont, is tarief 1 laag en tarief 2 hoog.

Diondk, 2021-02-24:
Voor de tarieven ging het mij er ook meer om dat op veel pagina's
laagtarief (tarief 1)
Hoogtarief (tarief 2) 

Staat weergegeven, dus leek me handig als het er ook zo uitzag op de energie contracten pagina.

dennissiemensma, 2021-02-24:
Bedankt voor je aanvulling. Vanwege de technische werking van het admin-portaal is dat dus wat lastig omdat het dynamisch is. Desnietemin kan ik er wel een algemene omschrijving voor NL-gebruikers toevoegen, dus dat zal ik dan doen ter verduidelijking.

[image]

dennissiemensma, 2021-02-24:
Extra decimaal komt ook mee in de volgende release. Vooral omdat het uiteindelijk minimale impact leek te hebben.

Daarbij de notitie voor dat het waarschijnlijk in de praktijk weinig tot niets gaat uitmaken, omdat overal in DSMR-reader dagtotalen gebruikt worden, die twee decimalen hanteren voor . Ook voor maanden, ook voor jaren. Dus in zeker zin zal er altijd een verschil zijn als het zelf nagerekend wordt.

closed 2021-02-24 by dennissiemensma

dennissiemensma, 2021-03-09:
Zojuist uitgebracht in [v4.13](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.13.0).