dsmrreader/dsmr-reader #1305: Schoonheidsfoutje op de statistieken pagina.

vincentweeda, 2021-02-22:
Beste,

Op de /statistics pagina is er een klein schoonheidsfoutje. 
De Elektriciteit (gecombineerd) kWh "mist" een label tag waardoor deze heel klein en wat slecht te lezen is. 
Zie ook onderstaande screenshot:
[image]

```

DSMR-READER
    App / Python / Database                                                       v4.12 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             134806
    Consumption records electricity / gas (est.)                                                  21358 / 319

```

dennissiemensma, 2021-02-22:
Bedankt voor je melding! Ik zal het bij een volgende release fixen.

vincentweeda, 2021-02-22:
Top! Ik zag trouwens op het dashboard nog zo'n geval alleen nu kan het zijn dat je dat expres gedaan had. 
[image]

dennissiemensma, 2021-02-22:
Ja klopt die op dashboard is expres, omdat er niet echt een kleur is voor dagen.

dennissiemensma, 2021-02-24:
Ik had trouwens niet gezien dat het over de statistieken ging, ik dacht het dashboard. 

Dan denk ik dat dit wel bewust is geweest destijds, omdat het geen "verbruik" is en ook geen echte meterstand, maar een optelling daarvan. Ik zal kijken of ik het toch wat duidelijker kan krijgen.

dennissiemensma, 2021-02-24:
Ik heb het aangepast voor de volgende release.

Op de pagina met contracten had ik ook al de badge eerder toegevoegd, dus daarom alsnog gewoon gelijkgetrokken bij de statistieken straks.

closed 2021-02-24 by dennissiemensma

dennissiemensma, 2021-03-09:
Zojuist uitgebracht in [v4.13](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.13.0).