dsmrreader/dsmr-reader #1307: (Opgelost) Error bij upgrade: ModuleNotFoundError: No module named 'setuptools._distutils'

voipmeister, 2021-02-25:
Bij de upgrade (non-docker) van 4.11 naar 4.12 ging de deploy niet goed in mijn installatie. Onderstaande foutmelding kwam voorbij:

```
ERROR: Exception:
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 228, in _main
    status = self.run(options, args)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 182, in wrapper
    return func(self, options, args)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 324, in run
    reqs, check_supported_wheels=not options.target_dir
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 183, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 388, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 340, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 483, in prepare_linked_requirement
    req, self.req_tracker, self.finder, self.build_isolation,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 91, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/wrappers.py", line 161, in get_requires_for_build_wheel
    'config_settings': config_settings
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/wrappers.py", line 265, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 86, in _build_backend
    obj = import_module(mod_path)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 6, in <module>
    import distutils.core
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/_distutils_hack/__init__.py", line 82, in create_module
    return importlib.import_module('._distutils', 'setuptools')
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'setuptools._distutils'
```

Een extra deploy verhielp dit niet. Uiteindelijk heb ik dit probleem opgelost door `export SETUPTOOLS_USE_DISTUTILS=stdlib` te doen en de deploy nogmaals te draaien. Hierna ging alles goed.

Issue aangemaakt ter logging, wellicht helpt het anderen.

voipmeister, 2021-02-25:
Issue aangemaakt ter logging, wellicht helpt het anderen.

closed 2021-02-25 by voipmeister

dennissiemensma, 2021-02-25:
Dank voor het delen! Weet je nog bij welke stap of command in het deployscript deze naar voren kwam?

dennissiemensma, 2021-03-25:
Inmiddels een soortgelijk issue gemeld in #1334, dus hopelijk lost je tip dat op :]