dsmrreader/dsmr-reader #1309: Oude gebruiker wordt niet verwijderd bij aanpassen via environment variables

LouisMT, 2021-02-25:
Gisteren heb ik DSMR Reader geïnstalleerd via Docker. Na het instellen besluit ik de standaard loginggevens (`admin`/`admin`) te wijzigen, voordat ik DSMR Reader openzet naar het internet. Dit doe ik door de environment variables `DSMRREADER_ADMIN_USER`  en `DSMRREADER_ADMIN_PASSWORD` in te stellen en de Docker te herstarten. De nieuwe login werkt, prima!

Maar nu is het mij opgevallen dat de oude gebruiker nog steeds in de database staat. Het is dus, naast mijn nieuwe "veilige" account, ook mogelijk om simpelweg in te loggen met `admin`/`admin`. Omdat ik dit heb geconfigureerd via de environment variables vind ik dit een beetje vreemd, bij alle andere Docker images die ik gebruik zouden de logingegevens juist vervangen worden.

Mijn voorstel zou zijn om een gebruiker die is geconfigureerd via environment variables eenmalig aan te maken, en vervolgens alleen de gebruikersnaam en het wachtwoord van deze gebruiker te updaten. Nieuwe gebruikers kunnen dan eventueel nog steeds worden aangemaakt via de CLI.

Ik heb dit issue niet in de Docker respository geplaatst omdat het in de "gewone" versie ook mogelijk is om met environment variables de gebruiker te beheren.

dennissiemensma, 2021-02-25:
Bedankt voor je melding. Dit lijkt me inderdaad een ongewenst bijeffect, dus ik zal ervoor zorgen dat alle bestaande admingebruikers verwijderd of uitgeschakeld worden bij uitvoeren van `dsmr_superuser` (wat de Docker image doet).

dennissiemensma, 2021-02-25:
Aangepast voor volgende release.

closed 2021-02-25 by dennissiemensma

dennissiemensma, 2021-03-09:
Zojuist uitgebracht in [v4.13](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.13.0).