dsmrreader/dsmr-reader #1312: Live verbruik http denial-of-service

GieltjE, 2021-03-03:
Het updaten van het live verbruik gebeurd ongeacht of de vorige update al afgerond is of niet, dit heeft tot gevolg dat als b.v. /live aangeroepen word de server onbedoeld een kleine denial-of-service te verwerken krijgt.

In dit voorbeeld zijn er genoeg apache worker threads beschikbaar, maar zodra het ophalen van de overige info lang genoeg duurt worden deze allemaal volgetrokken door de updates van het live verbruik die in de wachtrij staan.

[image]

```
        ProxyPass /static/ !
        ProxyPass / unix:/var/run/dsmr/gunicorn--dsmr_webinterface.socket|http://localhost/
```

dennissiemensma, 2021-03-03:
Bedankt voor je melding. Ik kan dat in een toekomstige release wel een keertje verbeteren. 

Hoelang duurt die XHR request nu uberhaupt in jouw situatie? Normaal gesproken is de performance van die request namelijk nooit een issue, omdat die queries relatief snel zouden moeten zijn.

GieltjE, 2021-03-03:
De XHR request voor het huidige gebruik is normaal een paar milliseconden (database load is practisch 0).
Alleen zodra je hem combineerd met de "live graphs" pagina (/live) gaat dit fout. De database heeft geen actieve queries (met mariadb zijn alle andere paginas milliseconden werk, zit genoeg rekenkracht in de machine). Alleen het ophalen van /live duurt gemiddeld ~30seconden (was ~45 met postgresql) welke dus het probleem veroorzaakt.

dennissiemensma, 2021-03-03:
Als de database vrijwel niets zwaars doet, zit het probleem mogelijk in de hoeveelheid data die geserialized wordt voordat het naar de browser gestuurd wordt. 

Kan het zijn dat je een zware combinatie hebt qua:
- Groeperen per meting ipv per minuut?
- Default zoom voor live grafieken relatief hoog (>10%)?
- Maximale range voor live grafieken (>24h)?

dennissiemensma, 2021-03-03:
Ik heb het ook even kunnen testen op mn Pi, met een variatie aan maximum live graph range van 7, 3 en 1 dagen:

### 7 dagen range
[image]

### 3 dagen range
[image]

### 1 dag range
[image]

----

Dit is overgens ook nog eens de meeste lichte versie, van groeperen per minuut. 

Als je kijkt naar je eigen screenshot zie bijvoorbeeld ook al een groot verschil qua data, gezien je XHR-request voor Ampere al 1,76 MB groot is. Als je inderdaad een grote range gebruikt daar, kun je nog overwegen die wat kleiner te maken.

Ik zou ooit een keer kunnen kijken of ik het omzetten van de DB-gegevens naar de JSON voor de browser kan omzetten naar een "streaming" response, waarbij die de data meteen al begint terug te geven, ook al is die nog bezig. Al weet ik niet of de grafieken library daar ook mee om kan gaan, maar mogelijk verkomt het ook die vage "locking".

dennissiemensma, 2021-03-03:
@GieltjE je zou ook nog kunnen proberen om het aantal "workers" van Gunicorn op te hogen. Gunicorn is het stukje software achter Nginx (of Apache in jouw geval) dat met de applicatie praat.

Het lijkt erop dat de huidige defaults van DSMR-reader slechts 1 worker gebruiken. Dat is wat minder handig.

Je kunt proberen om deze op te schakelen naar 4 (of meer) workers met `--workers 4` als extra argument op het proces dat de webinterface beheert:
```
command=/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--%(program_name)s.socket --pid /var/tmp/gunicorn--%(program_name)s.pid dsmrreader.wsgi
```

https://docs.gunicorn.org/en/latest/settings.html#worker-processes

Vergeet uiteraard niet het proces te herstarten.

Dit zal de requests niet sneller maken, maar waarschijnlijk wel voorkomen dat ze elkaar blokkeren.

dennissiemensma, 2021-03-03:
Het verhogen van de workers zorgt er bij mij iig voor dat ze elkaar niet meer blokkeren. Ze worden er dus niet sneller van, maar zitten elkaar niet meer in de weg. 

Afhankelijk van je eigen hardware en traffic kun je zelf het beste spelen met het optimale aantal workers.

dennissiemensma, 2021-03-03:
Ik zal voor de volgende release het aantal workers verhogen. Echter heeft dat alleen effect op nieuwe installaties, dus voor alle bestaande gebruikers zal ik het ook meenemen in de eerstvolgende major release.

GieltjE, 2021-03-03:
[image]

Denk dat ik inderdaad misschien wat veel datapunten heb, dit is met 8 workers, hij staat inderdaad op per reading zal dit eens aanpassen (moet er nog iets gebeuren nadat die setting is aangepast?).

dennissiemensma, 2021-03-03:
De grouping heeft alleen invloed op **nieuwe** datapunten. Het gaat je straks iig een factor 60 aan datapunten schelen per minuut.

----

Voor die workers moet je normaal Supervisor reloaden en herstarten, misschien werkt het in jouw geval met systemd iets anders. Hij moet iig de systemd config voor de webinterface herladen en het proces opnieuw starten, zodat die 8 workers spawnt.

Als dat goed gaat, zie je dat ook in de process list, 1 master process met (bij mij) 4 workers:
```
sudo ps faux | grep gunicorn
```

```
sudo ps faux | grep gunicorn
dsmr      4045  0.0  0.8  24540 16560 ?        S    19:54   0:02  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --workers 4 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
dsmr      4048  0.1  2.4  61548 46436 ?        S    19:54   0:09      \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --workers 4 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
dsmr      4049  0.0  2.4  62612 47368 ?        S    19:54   0:06      \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --workers 4 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
dsmr      4050  0.1  2.3  59420 44012 ?        S    19:54   0:09      \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --workers 4 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
dsmr      4051  0.0  2.2  58596 43204 ?        S    19:54   0:06      \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --workers 4 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi

```

GieltjE, 2021-03-03:
Om de een of andere reden accepteerd systemd geen override van de ExecStart, dus de vorige test was nog met 1 worker.
Met 4 daadwerkelijke workers ziet het er inderdaad een stuk beter uit:
[image]

dennissiemensma, 2021-03-03:
Oke mooi! Waarschijnlijk zie je verder door het groeperen vanaf morgen ook al een flinke vermindering in de datagrootte en bijbehorende laadtijd van de grafieken daar.

dennissiemensma, 2021-03-03:
Ik heb de standaardconfig voor nieuwe installaties nu ook aangepast naar 4 workers, dus dat gaat vanzelf mee met een volgende release.

closed 2021-03-03 by dennissiemensma

dennissiemensma, 2021-03-06:
@GieltjE heeft het groeperen van de metingen het aantal datapunten/datagrootte in de responses verkleint? En zie je dat ook terug in de responsetijd?

GieltjE, 2021-03-07:
@dennissiemensma Ja, scheelt best wel significant veel tijd:
[image]

dennissiemensma, 2021-03-07:
Top, fijn om te horen!

dennissiemensma, 2021-03-24:
In de nieuwe release heb ik het overigens nog wat verder gerefactored zodat er minder of geen nieuwe XHR-requests gedaan worden als (dezelfde) nog onderweg is. Dat maakt het ook nog wat soepeler in de toekomst.