dsmrreader/dsmr-reader #1314: Drop support for Python 3.6 (EOL December 2021)

dennissiemensma, 2021-03-03:
It will reach end of life December 2021: https://devguide.python.org/#status-of-python-branches

However, since 3.7, 3.8 and 3.9 have been around for a while, it should be okay to require 3.7+ in the next major DSMR-reader release.

dennissiemensma, 2021-03-09:
Deprecation uitgebracht in [v4.13](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.13.0).

dajappie, 2021-03-18:
Due to deprecation of Python 3.6 I've upgraded to Python 3.9.2 on my Rpi 3b (stretch). When then upgrading from DSMRreader 4.12 to 4.13 there were errors about Python cryptography. They did not seem to interfere with DSMRreader running when ignored, but to entirely resolve them I had to install Rust before the Python crypto module could be installed. Below commands are sufficient to get rid of all crypto errors during the DSMR upgrade:

```
sudo su - dsmr
pip install --upgrade pip
curl https://sh.rustup.rs -sSf | sh
. ~/.profile
pip install cryptography
./deploy.sh
```

dennissiemensma, 2021-03-18:
@dajappie bedankt voor je aanvulling. Had je ook de complete virtualenv opnieuw aangemaakt in Python 3.9?

dajappie, 2021-03-26:
Sorry voor de late reply, ja Python nieuw van scratch gecompileerd, oude virtualenv weggegooid en nieuwe aangemaakt. Upgrade naar laatste versie liep gisteren probleemloos met de crypto issue opgelost

dennissiemensma, 2021-03-26:
Fijn, dank voor de update! Goed om te weten dat dit dan mogelijk een dingetje gaat worden bij andere gebruikers tzt.

dennissiemensma, 2021-10-21:
Tevens ondersteuning voor Python 3.10 toegevoegd. Die is begin deze maand uitgebracht en de tests in DSMR-reader lijken er niet door stuk te gaan.

closed 2021-10-21 by dennissiemensma