dsmrreader/dsmr-reader #1315: Rework documentation structure

dennissiemensma, 2021-03-06:
Momenteel is de documentatie redelijk uitgewerkt. Echter, ik merk dat het moeilijk is om er een goede structuur in te maken. Het is namelijk deels promotie van features, deels FAQ/zelfhulp, deels handleidingen en ik merk dat het soms wat lastig is om daar een natuurlijke indeling in te maken.

Een tijdje terug kwam ik deze tegen en dat is superinteressant, namelijk gewoon de documentatie splitsen in vier categorieen:
https://documentation.divio.com/

- Tutorials
- How-to
- Explanation
- Reference

Moeite waard om wat tijd in te steken. Ik denk dat het uiteindelijk de gebruikers goed kan helpen.

jeroenpeters1986, 2021-03-06:
Zeker de moeite waard, ik gebruik dat veel voor DjangoCMS en als
gebruiker/lezer is dat heel erg fijn!
-- 
Groeten,

Jeroen Peters

dennissiemensma, 2021-03-14:
Eerste opzet staat. Zal wel wat iteraties nodig hebben voordat het optimaal is.

commit 97cdfec2 (2021-03-14): Doc translation fix #1315
commit 53fea6a5 (2021-03-14): Doc translation fix #1315
commit 03575fd4 (2021-03-14): Doc translation fix #1315
closed 2021-03-14 by dennissiemensma