dsmrreader/dsmr-reader #1316: Documentatie bijwerken voor het ontvangen van telegrammen.

erikkral, 2021-03-06:
Hallo,

Ik maak al enige tijd gebruik van DSMR-reader. Zelf gebruik ik een ESP8266 die met ESP easy bij mijn slimme meter. Met een Python scriptje op mijn server haal ik de telegrammen op uit ESPEasy en worden die doorgestuurd naar de API van DSMR reader. Had er zelf al eens een topic voor geopend om dit draaiende te krijgen.

Ik zou eigenlijk van dit python scriptje af willen en het op een meer robuuste manier willen inrichten. Dacht eraan om dit via MQTT te doen. Heb toen MQTT in DSMR reader geconfigureerd maar toen liep ik vast op het hoe en wat. Ik kwam er toen achter dat de MQTT in DSMR reader alleen gebruikt kan worden voor het verzenden van data.

Mijn suggesties zitten hem een klein beetje in de functionaliteit, namelijk via MQTT een bericht kunnen ontvangen maar des te meer zie ik een verbeterpunt in de documentatie. De functies van DSMR reader zijn vrij goed gedocumenteerd maar is lastig te begrijpen hoe je het een en ander functioneel kan inzetten. De MQTT broker kan dus wel informatie verzenden (waarom zou je dat doen trouwens?) en niet ontvangen. De API kan wel informatie ontvangen, maar daar moet je dan zelf iets voor inrichten, in mijn geval een python scriptje wat telegrammen doorstuurt. Maar misschien kan de API ook wel gegevens versturen naast het ontvangen. Ik weet het niet.

De voorbeelden met MQTT en API zijn niet om te zeuren dat er iets mis gaat of iets ontbreekt. Ik probeer duidelijk te maken dat de documentatie goed is maar dat deze wat moeilijk te plaatsen is voor een DSMR Reader beginner.
Zouden we de documentatie iets kunnen verbeteren zodat een beginner ook snapt wat er wel en niet kan? 

Misschien zou er meer per use case gedocumenteerd kunnen worden. Dus dat er een situatie beschreven wordt van DSMR Reader die draait op een RPi die bij de meter staat, een situatie waarbij DSMR Reader op een server staat en de telegrammen toegestuurd krijgt vanaf een esp8266 (bijvoorbeeld) enzovoorts...

Bij dit idee hoort ook mijn aanbod voor hulp. Ik kan best helpen met documenteren, als anderen hier ook brood in zien.

dennissiemensma, 2021-03-06:
Bedankt voor je verzoek. Het lijkt me een goede aanvulling om daar wat meer informatie over uit te werken. Via #1315 wil ik eerst de documentatie wat duidelijker opdelen en daarna zou ik nog dit soort aanvullingen kunnen doen.

dennissiemensma, 2021-04-01:
Ik heb een klein beetje documentatie toegevoegd voor de volgende release. 

Toch denk/merk ik dat heel veel gebruikers de documentatie alsnog niet (eerst) lezen, dus heb ik het beperkt gehouden. Ik kan dan beter verwijzen naar oude issues.

closed 2021-04-01 by dennissiemensma

dennissiemensma, 2021-04-03:
Onderdeel van de nieuwe [v4.15 release](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.0).