dsmrreader/dsmr-reader #1317: High / Low tariff depending on location

arjenfvellinga, 2021-03-07:
As I understand from https://www.enexis.nl/consument/aansluiting-en-meter/tarief/dubbeltarief the hours for the high and low tariff depend on where you are living. But I can't find a setting to set you location. So I don't know now if the number of low and high tariff is calculated correctly.

Is this a missing settings, or am I missing something?

dennissiemensma, 2021-03-07:
Bedankt voor je vraag. Als het goed is **regelt je slimme meter dit zelf**. Deze weet exact welk tarief er op het moment geldt en houdt daar (vier) losse meterstanden voor bij. 

Vandaag is dit bijvoorbeeld goed te zien, want er geldt in het weekend alleen laagtarief:

[image]

In je eigen telegram kun je het ook zien aan deze velden:

[image]

De bovenste vier zijn:
- Verbruikt tarief 1
- Teruggeleverd tarief 1
- Verbruikt tarief 2
- Teruggeleverd tarief 2

En de onderste:
- Huidig tarief (1)

In Nederland is tarief 1 laag en tarief 2 hoog.

arjenfvellinga, 2021-03-07:
Ah duidelijk!

Dank voor je snelle antwoord!

closed 2021-03-07 by arjenfvellinga