dsmrreader/dsmr-reader #1318: backups naar synology?

gobartos, 2021-03-07:
Ik heb DSMR-reader op dezelfde VM als Home Assistant geïnstalleerd, met Proxmox als host.
Zoals je merkt ken ik weinig van Linux, dus mijn excuses. Maar op welke manier kan ik de backup naar mijn Synology maken? Die beschikt over Linux indeling indien gewenst.

dennissiemensma, 2021-03-07:
Bedankt voor je vraag. Ik kan je hier helaas niet mee helpen en alleen doorverwijzen. Je kunt de link in #701 proberen en anders ben je denk ik het het beste geholpen op een Synology-forum, HA-forum of het Tweakers-forum. 

In de DSMR-reader admin kun je iig een map opgeven waar de backup's heen gaan. Standaard is dat `backups/`, gezien vanaf `/home/dsmr/dsmr-reader/`, dus absoluut pad is `/home/dsmr/dsmr-reader/backups/`. 
Je kunt ook (indien nodig) een ander, absoluut pad opgeven, mits de `dsmr` user op je VM naar dat pad kan schrijven. 

Het uitzoekwerk zit hem dan nog voor je in om bovenstaande map periodiek naar je NAS te krijgen.

closed 2021-03-07 by dennissiemensma

gobartos, 2021-03-07:
Bedankt, dat is alvast een duwtje in de goeie richting. Ik vrees dat er toch erg veel data op mij af komt :)