dsmrreader/dsmr-reader #1319: APIv2 returns numbers as strings

Proxx, 2021-03-09:
Is it possible to change the datatype from string to numbers in the api?

API call: {{DSMR-reader-API-host}}/api/v2/statistics/day

``` json
{
    "count": 1,
    "next": null,
    "previous": null,
    "results": [
        {
            "id": 358,
            "day": "2021-02-28",
            "total_cost": "4.46",
            "electricity1": "6.163",
            "electricity2": "0.000",
            "electricity1_returned": "0.000",
            "electricity2_returned": "0.000",
            "electricity1_cost": "1.26",
            "electricity2_cost": "0.00",
            "gas": "4.383",
            "gas_cost": "3.20",
            "lowest_temperature": "0.0",
            "highest_temperature": "0.0",
            "average_temperature": "0.0",
            "fixed_cost": "0.00"
        }
    ]
}
```

should be returned as 
``` json
{
    "count": 1,
    "next": null,
    "previous": null,
    "results": [
        {
            "id": 358,
            "day": "2021-02-28",
            "total_cost": 4.46,
            "electricity1": 6.163,
            "electricity2": 0.000,
            "electricity1_returned": 0.000,
            "electricity2_returned": 0.000,
            "electricity1_cost": 1.26,
            "electricity2_cost": 0.00,
            "gas": 4.383,
            "gas_cost": 3.20,
            "lowest_temperature": 0.0,
            "highest_temperature": 0.0,
            "average_temperature": 0.0,
            "fixed_cost": 0.00
        }
    ]
}
```

dennissiemensma, 2021-03-09:
Bedankt voor je melding! Dit is mij nog niet eerder opgevallen en ook zeker niet de bedoeling.

Ik gebruik er een redelijk standaard API framework voor, samen een standaard-serializer, die de velden van het betreffende model zou moeten pakken. In dit geval voornamelijk Decimals waarbij ik een conversie naar JSON floats zou verwachten. 
Ik zie echter  niet zo 1-2-3 de oorzaak ervan, dus ik kijk er in een toekomstige release naar. Voor nu kun je ze het beste client-side typecasten naar de juiste type(s).

dennissiemensma, 2021-03-09:
En omdat de toekomstige fix dan ook eens backwards-incompatible is, voor clients die wel uitgaan van de strings zoals ze nu zijn, kan ik het alleen meenemen bij een major release. 
Dus dat zal wel even duren, gezien er ook een hoop andere wijzigingen daarvoor op de planning staan.

dennissiemensma, 2021-04-07:
Ik weet niet of het altijd al zo was, of sinds een upgrade bij Django REST Framework, maar hier staat [een soortgelijk issue](https://github.com/encode/django-rest-framework/issues/2417). Ik kan het vrij eenvoudig fixen door `COERCE_DECIMAL_TO_STRING=False` te gebruiken in de framework-instellen.

Echter, het lijkt erop dat Decimals bewust als strings worden teruggegeven ivm de nauwkeurigheid, omdat computers issues kunnen hebben met het parsen van floats. Dat is ook een van de redenen dat ik interne Python Decimals gebruik, omdat die ervoor gemaakt zijn.

Dus nu twijfel ik wat verstandig is. Ook hier staat een [betoog](https://stackoverflow.com/a/38357877) om juist strings te behouden en het parsen aan de clients over te laten. Ik denk dat ik het daarom maar laat voor wat het is. Het is visueel niet mooi, maar er zit dus een goede reden achter die ik zelf even vergeten was.

dennissiemensma, 2021-04-07:
@Proxx desnietemin kun je ervoor kiezen om het lokaal wel anders in te stellen als je dat heel graag wilt. 

Het is een custom override voor puur je eigen installatie dan, maar voel je vooral vrij om het toe te passen als je dat wenselijk vindt.

In `dsmrreader/settings.py` onderaan een regel toevoegen met:
```
REST_FRAMEWORK['COERCE_DECIMAL_TO_STRING'] = False
```

En dan de applicatie (hier met name het `dsmr_webinterface` proces) herstarten. Bijvoorbeeld:

```
sudo supervisorctl restart all
```

dennissiemensma, 2021-04-07:
Binnen DSMR-reader houd ik het dus standaard toch op strings n.a.v. bovenstaande.

closed 2021-04-07 by dennissiemensma