dsmrreader/dsmr-reader #1320: Dashboard op iPhone XS

vincedh, 2021-03-10:
Hoi Dennis,

Sinds v4.13 past het dashboard niet meer op m'n iPhone XS, zie bijlage.

Bedankt weer!

[image]

dennissiemensma, 2021-03-10:
Bedankt voor je melding. Ik ondersteun heel beperkt mobiele devices, want er is geen beginnen aan om ze allemaal telkens te gaan controleren en ondersteunen.

Voor de volgende release heb ik een algemene verbetering klaarstaan. Mogelijk lost dat ook jouw situatie op.

closed 2021-03-10 by dennissiemensma

vincedh, 2021-03-10:
Super, bedankt weer voor je snelle reactie, we gaan het zien :)

dennissiemensma, 2021-03-10:
Je kunt de fix trouwens ook zelf doorvoeren tot die tijd.

Waar je DSMR-reader geinstalleerd hebt, dit bestand openen:
```
/var/www/dsmrreader/static/dsmr_frontend/css/dsmr-reader/global.css
```

En dan onderaan deze code toevoegen:
```
@media screen and (max-width: 560px) {
  table.table {
    font-size: 10px !important;
  }
  table span.badge {
    font-size: 10px !important;
  }
}
```

Zoals hier rechts: https://github.com/dsmrreader/dsmr-reader/commit/6eca22cc83f7d5d3bbca67c7a8a92348aa8f08e3#diff-1132742a27a58f25733bd11198d29e6522ce6bec125c4a9bb52edfb1794ec8cf

Daarna in je iOS browser een harde refresh doen van het dashboard.

vincedh, 2021-03-10:
Dat heeft absoluut geholpen! Tnx!

dennissiemensma, 2021-03-23:
Uitgebracht in [v4.14](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.14.0)