dsmrreader/dsmr-reader #1322: Opnieuw genereren van dagstatistieken gebeurd niet voor Gas

nsvdbosch, 2021-03-12:
Ik heb hier mogelijk wat voor kunnen maken. In de volgende v4.13 release kun je dit proberen:

```
sudo su - dsmr
./manage.py dsmr_stats_reconstruct_missing_day_statistics
```

_Originally posted by @dennissiemensma in https://github.com/dsmrreader/dsmr-reader/issues/1302#issuecomment-792328271_

dennissiemensma, 2021-03-12:
Bedankt voor je melding. Heb je toevallig ook een `debuginfo`-dump van de About-pagina? 

Gas is optioneel en het hangt er vanaf of/hoe je die geimporteerd hebt. Er zou namelijk geen onderscheid gemaakt moeten worden, maar gas [wordt wel geskipped als er niets gevonden wordt](https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_stats/services.py#L132-L158) voor die dag.

nsvdbosch, 2021-03-12:
> Bedankt voor je melding. Heb je toevallig ook een `debuginfo`-dump van de About-pagina?
> 
````

DSMR-READER
    App / Python / Database                                                       v4.13 / v3.8.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 5.0s / 4032h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             544592
    Consumption records electricity / gas (est.)                                                 95423 / 4140

````

dennissiemensma, 2021-03-12:
Je zou nog kunnen proberen om:
- [debug-logging](https://dsmr-reader.readthedocs.io/en/v4/faq.html#logging-levels) tijdelijk in te schakelen
- daarna de database-tabellen met uur & dagstatistieken te legen (`dsmr_stats_hourstatistics` / `dsmr_stats_daystatistics`)
- en dan het reconstruct-commando opnieuw uitvoeren.

Maak voor de zekerheid wel even een backup vooraf.

Wellicht werkt dat beter en je ziet dan mogelijk ook meer info door de debug-log. Als dat het niet oplost of een andere oorzaak geeft, kan ik het wel een keertje voor je proberen in mijn eigen omgeving en zoeken naar de oorzaak.

nsvdbosch, 2021-03-15:
> Je zou nog kunnen proberen om:
>
> * [debug-logging](https://dsmr-reader.readthedocs.io/en/v4/faq.html#logging-levels) tijdelijk in te schakelen
> * daarna de database-tabellen met uur & dagstatistieken [...]

Bovenstaande acties uitgevoerd. Helaas zonder het gehoopte resultaat en geen enkel aanknopingspunt vanuit de debugging. Wat heb je nodig voor je eigen omgeving. een backup?

dennissiemensma, 2021-03-15:
Het makkelijkste is om dan idd een backup te maken van alleen de twee tabellen die je gevuld hebt:

```
sudo su -postgres
pg_dump -d dsmrreader -t dsmr_consumption_electricityconsumption -t dsmr_consumption_gasconsumption --data-only | gzip --fast > dump.sql.gz
```

Tenminste, ik ga er vanuit dat je daar de gegevens in gezet hebt. En die mag je dan mailen naar github at dennissiemensma punt nl.

dennissiemensma, 2021-03-16:
Bedankt voor je dump. Het lijkt erop dat je gasmetingen net een uur scheef staan.

Je kunt het wel zien als je deze query uitvoert, met onze tijdzone als weergave:

```
sudo su -postgres
psql dsmrreader
```

```
select read_at at time zone 'Europe/Amsterdam', delivered
from dsmr_consumption_gasconsumption
where read_at < '2021-01-05 00:00:00.000000 +01:00'
order by read_at asc;
```

Je ziet dan bijvoorbeeld dit:
```
2016-11-01 22:59:00.000000   95.287
2016-11-01 23:00:00.000000   95.287

2016-11-02 22:59:00.000000   104.420
2016-11-02 23:00:00.000000   104.420
```

Dit is in onze tijdzone en DSMR-reader gebruikt dat ook en kijkt naar de eerste en laatste meting van een dag. Omdat ze scheeft staan is de beginstand van je oude data ook altijd de eindstand en komt er dus altijd `0` verschil/verbruik uit.

Als ik dezelfde query opnieuw uitvoer, maar dan in een tijdzone later (1 uur later dan in NL, bijvoorbeeld Griekenland), dan zie je dat het wel goed gaat en dat is waarschijnlijk wat de oorspronkelijke bedoeling was:
```
select read_at at time zone 'Europe/Athens', delivered
from dsmr_consumption_gasconsumption
where read_at < '2021-01-05 00:00:00.000000 +01:00'
order by read_at asc;
```

```
2016-11-01 00:00:00.000000   88.654
2016-11-01 23:59:00.000000   95.287

2016-11-02 00:00:00.000000   95.287
2016-11-02 23:59:00.000000   104.420
```

Nu schuift alles goed door.

Om het oplossen kun je ofwel al die data opnieuw proberen te importeren met de goede tijdzone. Ofwel proberen het in bulk recht te trekken. 

Dat laatste was ik al aan het uitzoeken, maar dat schijnt niet zo makkelijker te zijn in dit geval, omdat Postgres niet zomaar alles + 1 uur kan doen, gezien de zomertijd/wintertijd wisselingen daar roet in het eten gooien. Ik zal nog even verder zoeken.

----

Het **kan** trouwens zijn dat hetzelfde geldt voor je al oude elektra stats, maar dat kan ik niet zien. Daarbij **kan** het dus zijn dat alle dagtotalen net een uur zijn verschoven. 
Waarschijnlijk zorgt dat niet voor schokkende verschillen, maar je wilt het wellicht even dubbelchecken als je nog ergens een oude referentie hebt ter vergelijking met je oude elektra dagtotalen in DSMR-reader.

dennissiemensma, 2021-03-16:
Uiteindelijk toch een bulk fix voor gas gevonden:

```
sudo su -postgres
psql dsmrreader
```

```
update dsmr_consumption_gasconsumption
set read_at = read_at + interval '1 hour'
where read_at < '2021-01-05 00:00:00.000000 +01:00';
```

Maak sowieso even een backup voor de zekerheid en dubbelcheck even mijn aanname dat al je geimporteerde data van voor 5 januari is: `'2021-01-05 00:00:00.000000 +01:00'`.

----

Daarna kun je deze weer proberen, als het goed is zijn de gasmetingen nu goed qua start/eind per dag. Check dat even met:
```
select read_at at time zone 'Europe/Amsterdam', delivered
from dsmr_consumption_gasconsumption
where read_at < '2021-01-05 00:00:00.000000 +01:00'
order by read_at asc;
```

Advies is om debug logging uit te schakelen voordat je onderstaande doet. Gezien het ontzettend veel logt.

Tot slot kun je dit nogmaals doen:
- daarna de database-tabellen met uur & dagstatistieken te legen (`dsmr_stats_hourstatistics` / `dsmr_stats_daystatistics`)
- en dan het reconstruct-commando opnieuw uitvoeren.

nsvdbosch, 2021-03-17:
Super bedankt! Probleem is inderdaad ontstaan door het verschil in UTC en CET. Nu staan alle gegevens voor het gas correct.

closed 2021-03-17 by dennissiemensma